ARMmbed/mbed-os #12788: CAN: Remote frame is not working for H743ZI2

kpyush, 2020-04-10:
### Remote Frame is not working in CAN api

#### Target(s) affected by this defect ?
TARGET_NUCLEO_H743ZI2

#### Toolchain(s) (name and version) displaying this defect ?
GNU Arm Embedded Toolchain V7.2.1

#### What version of Mbed-os are you using (tag or sha) ?
mbed-os-5.14.1

#### How is this defect reproduced ?
Observe only printf("Data: %d, Type:CANData\n", msg.data[0]); is being printed.
```
#include "mbed.h"

Ticker ticker;

DigitalOut led1(LED1);
DigitalOut led2(LED2);

CAN can1(MBED_CONF_APP_CAN1_RD, MBED_CONF_APP_CAN1_TD);
CAN can2(MBED_CONF_APP_CAN2_RD, MBED_CONF_APP_CAN2_TD);
char counter = 0;

void send() {
  CANMessage msg;
  msg.id = 123;
  msg.type = CANRemote;
  msg.format = CANStandard;
  msg.len = 1;
  msg.data[0] = counter;
  if (can1.write(msg)) {
    counter++;
    printf("Message sent: %d\n", counter);
  }
  led1 = !led1;
}

int main() {
  ticker.attach( & send, 1);
  CANMessage msg;
  while (1) {
    if (can2.read(msg)) {
      if (msg.type == CANRemote) {
        printf("Data: %d, Type:CANRemote\n", msg.data[0]);
      } else {
        printf("Data: %d, Type:CANData\n", msg.data[0]);
      }
      led2 = !led2;
    }
    wait(0.2);
  }
}

```

jeromecoutant, 2020-04-10:
Hi

What is your setup? 2 boards connected with 2 external CAN shield ?
What are mbed_app.json values?
Is your setup working with other targets?

Thx

kpyush, 2020-04-10:
[image]
Setup is something like above image, communication is happening between two onboard can modules (CAN1 and CAN2 )of  [Nucleo H743zi](https://www.st.com/en/evaluation-tools/nucleo-h743zi.html) board. **Everything works fine as I am able to test filters, modes and formats** .. only remote frame is not working.

kpyush, 2020-04-10:
@ciarmcom I've updated the description and required fields.

jeromecoutant, 2021-06-28:
Hi @kpyush 
We think issue is corrected since #13565
Let's close this one ?
Thx

MubeenHCLite, 2021-08-03:
@kpyush 

A reminder to close th issue as this was resolved.

kpyush, 2021-08-03:
Closing the issue.. thanks.

closed 2021-08-03 by kpyush