ARMmbed/mbed-os #12789: I2C initialization on STM32 may be deviating from the norm

ganzuul, 2020-04-11:
### Description of defect

I2C on STM32 may be deviating from the norm

#### Target(s) affected by this defect ?
STM32

#### Toolchain(s) (name and version) displaying this defect ?

PlatformIO, version 4.3.1

#### What version of Mbed-os are you using (tag or sha) ?

5.14.1

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

#### How is this defect reproduced ? 
The I2C example;

```
class I2CPreInit : public I2C
{
public:
    I2CPreInit(PinName sda, PinName scl) : I2C(sda, scl)
    {
        frequency(400000);
        start();
    };
};
```
from  https://os.mbed.com/users/nkhorman/code/Adafruit_GFX/
works without `start();`, but not with it.  I only have STM32 devices to test this with.

The maintainer of that lib has not been active since 2014, but it is very popular, and I'm not sure how to document my finding of this simple fix.

0xc0170, 2020-04-15:
cc @ARMmbed/team-st-mcd

0xc0170, 2020-04-15:
@ganzuul have you tried any other sensor ? have you analyzed why that start condition is there (using logic analyzer) , what is being on  the bus? 

I don't think you need that preinit there. Similar to what others did (https://os.mbed.com/users/frada/code/SSD1306_demo//file/7a3182a3fa21/main.cpp/ )

0xc0170, 2020-04-15:
What exact target you have?

jeromecoutant, 2020-04-20:
> #### What version of Mbed-os are you using (tag or sha) ?
> 5.14.1

For us, it is a very old version....
Is it possible to try more recent branch ?
Thx

0xc0170, 2020-04-27:
I am closing this issue due to inactivity

closed 2020-04-27 by 0xc0170

ganzuul, 2020-04-28:
Very sorry about the inactivity! I did not get notifications for the replies. I don't think this needs to be re-opened, but I'll still reply to the questions.

While testing this just now I used a minimal project that I had used before, but now I'm suddenly unable to replicate the issue. I2C::start() causes no issue in the minimal project though I could have sworn it had the same issue before. In my main project the issue with I2C::start() remains. 

My target is seeedArchMax but I'm actually using a generic STM32F407VET6 board.

Platform IO reports:
`Updating framework-mbed                  @ 5.51103.190208 [Up-to-date]`
"51103.190208" is uninformative to me. 

My I2C device is an OLED panel without a reset pin. I might have some I2C sensors that I could try in the unlikely event that we return to this issue. (Again, I'm just attempting to create a bit of documentation about what worked for me.)

The I2CPreinit is used to set the clock rate of the channel and it defaults to 100 kilobaud. Without mbed having DMA for I2C getting the bus speed up seems important enough in general to make just a little noise about. 

With start() commented out:
[image]
Not very clean edges but works a treat.

With start() still commented out. A longer timescale. Nothing is shown on the display for all this activity but I could use the display normally in this state.
[image]

With start() in place. Longer timescale still. On reset it pulls both pins high and then flatlines. No clock and no data. 
[image]

LMESTM, 2020-04-29:
The start command actually send 'manually' an I2C START
https://github.com/ARMmbed/mbed-os/blob/master/drivers/I2C.h#L173
It should be followed by an address, then read and write **unitary** commands and a STOP.

If you're using more advanced APIs for I2C with address and multiple bytes, then the drivers takes care of sending the start and stop commands ... 
https://github.com/ARMmbed/mbed-os/blob/master/drivers/I2C.h#L158

Maybe your driver is trying to mix-up both set of commands ?