ARMmbed/mbed-os #12793: TARGET_STM32F769xI: incorrect vector table length

tdjastrzebski, 2020-04-13:
### Description of defect
Vector table length (**114**) for STM32F769 defined in file [cmsis_nvic.h](https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_STM/TARGET_STM32F7/TARGET_STM32F769xI/device/cmsis_nvic.h) - b97ffe8 appears to be incorrect.  
Correct vector table length for STM32F769 is **126** (16 core + 110 peripherals). Refer to [startup_stm32f769xx.S](https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_STM/TARGET_STM32F7/TARGET_STM32F769xI/device/TOOLCHAIN_GCC_ARM/startup_stm32f769xx.S).  

As a result dependent link script (e.g. [STM32F769xI.ld](mbed-os/targets/TARGET_STM/TARGET_STM32F7/TARGET_STM32F769xI/device/TOOLCHAIN_GCC_ARM/STM32F769xI.ld)) is built with incorrect values and program may crash.

### Target(s) affected by this defect ?
STM32F769xI

### Toolchain(s)
All using **NVIC_NUM_VECTORS** variable defined in **cmsis_nvic.h**: GCC_ARM, IAR, ARM_STD.  
ARM_MICRO has correct vector table length hardcoded.

### What version of Mbed-os are you using (tag or sha) ?
latest

0xc0170, 2020-04-15:
cc @ARMmbed/team-st-mcd

jeromecoutant, 2020-04-15:
I agree with the issue

Thx @tdjastrzebski for finding it!

PR #12860 (2020-04-28): STM32F7 update drivers version to CUBE V1.16.0
### Summary of changes 

ST Cube drivers version is updated from V1.15.0 to V1.16.0,
which is the latest official version from:
https://github.com/STMicroelectronics/STM32CubeF7

Fix #12793

#### Impact of changes 

#### Migration actions required 

### Documentation 

----------------------------------------------------------------------------------------------------------------
### Pull request type 

    [x] Patch update (Bug fix / Target update / Docs update / Test update / Refactor)
    [] Feature update (Ne
closed 2020-04-28 by 0xc0170