ARMmbed/mbed-os #12794: Mbed deploy fails on Windows and Linux due to Python modules installation failure

srikanthds, 2020-04-13:
### Description of defect

The `mbed deploy` fails on Windows with the following error. Likely due to unsupported version of *pywin32* (v224) and an old version of *cryptography* (v2.5) is listed in *requirements.txt* file. The *pip* is expecting the minimum version *pywin32* to be 225 and the *cryptography* version to be 2.8 (which has wheels) for successful installation.

```
[mbed-8284] ERROR: Missing Python modules were not auto-installed.
       The Mbed OS tools in this program require the following Python modules: mbed_cloud_sdk, jinja2, mbed_ls, mbed_host_tests, mbed_greentea, manifest_tool, icetea, pycryptodome, hidapi, wmi, psutil
       You can install all missing modules by running "pip install -r requirements.txt" in "C:\Users\srds\Downloads\mbed-os-example-template\mbed-os"
```

It fails on Linux if the **setuptools** version is old (say v20.7.0). The _requirements.txt_ file does not specify the *setuptools* version which cause the failure.

#### Target(s) affected by this defect ?

All

#### Toolchain(s) (name and version) displaying this defect ?

All

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-5.15.1

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

* mbed-cli 1.10.3
* Python 3.8.2

#### How is this defect reproduced ?

* Create a Python virtual environment
* Install mbed-cli
* Clone any Mbed OS example
* Configure the project root and run `mbed deploy`
* Observe that deploy fails on Windows and Linux

0xc0170, 2020-04-27:
https://github.com/ARMmbed/mbed-os/issues/12622 - There are at least 2 similar reports. Please use python 3.7 for now.

> It fails on Linux if the setuptools version is old (say v20.7.0). The requirements.txt file does not specify the setuptools version which cause the failure.

What failure is reported?

0xc0170, 2020-04-27:
cc @ARMmbed/mbed-os-tools