ARMmbed/mbed-os #12796: CAN Modes needs more elaborate documentation.

kpyush, 2020-04-13:
### Description of defect
Please provide the documentation around the different modes in CAN 

#### Target(s) affected by this defect ?
TARGET_NUCLEO_H743ZI2

#### Toolchain(s) (name and version) displaying this defect ?
GNU Arm Embedded Toolchain V7.2.1

#### What version of Mbed-os are you using (tag or sha) ?
mbed-os-5.14.1

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
NA

#### How is this defect reproduced ? 
NA

AGlass0fMilk, 2020-04-22:
I agree that Mbed's CAN API is not very well documented nor exemplified.

You can have a look at a project I am working on here (OBDII over CAN w/ Mbed) to see how I used filters:
https://github.com/AGlass0fMilk/mbed-OBD2/blob/master/OBD2.cpp

adbridge, 2020-05-05:
@kpyush please raise this in our forums. Github is for defects only. Many thanks.

closed 2020-05-05 by adbridge