ARMmbed/mbed-os #12798: [WI-SUN] Network loses stored certificates after first connect

felipeLeast, 2020-04-13:
## Description of defect

I've developed a routine in my program that check's for internet connectivity. When some time goes by without response I reset my wi-sun connection, the problem I'm facing is that when I attepmt to disconnect and connect again the WisunInterface on the authentication process I get "no own certificate", after some debuging I've found that the problem lies on the fact that after the certificates load stored certificates from `wisun_tasklet_add_stored_certificates` it deletes from the list of certificates.

I can see two options to fix this problem:
1- Comment line 498 from wisun_tasklet.c which deletes stored certificates
2- On disconnect delete em free `wisun_tasklet_data_ptr`, which gives me the possibility to reload my certificates.

#### Target(s) affected by this defect ?
K64F

#### Toolchain(s) (name and version) displaying this defect ?
ARM_GCC 

#### What version of Mbed-os are you using (tag or sha) ?
mbed-os 5.15.1

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
mbed-cli 1.10

#### How is this defect reproduced ? 
On mbed-os-example-mesh-minimal add after:
`printf("Connected. IP = %s\n", sockAddr.get_ip_address());`
```
mesh->disconnect();
mesh->connect();
while (NSAPI_ERROR_OK != mesh->get_ip_address(&sockAddr))
    ThisThread::sleep_for(500);
printf("Connected. IP = %s\n", sockAddr.get_ip_address());
```

0xc0170, 2020-04-14:
cc @artokin

felipeLeast, 2020-05-15:
Hi @artokin, any comments regarding this issue?

eriknayan, 2020-05-15:
I have the same problem here with an STM32WB55 custom board.

I tried removing the certificates and setting them again, but still after the Network Interface disconnects and reconnects back, I get "no own cert" error.

mikter, 2020-05-25:
We will start working on this issue sorry for the delay.

mikter, 2020-06-01:
@felipeLeast and @eriknayan I created pull request to fix this could you check that is fixes the issue you are having

felipeLeast, 2020-06-08:
PR Fixed issue!

closed 2020-06-08 by felipeLeast