ARMmbed/mbed-os #12800: Update cmsis pack index.json partially

ccli8, 2020-04-14:
### Description of defect

I am porting Nuvoton's new target onto Mbed OS. To enable bootloader and export to Keil and IAR, I need to update cmsis pack information `tools/arm_pack_manager/index.json`. According to [Porting a new target for exporters](https://os.mbed.com/docs/mbed-os/v6.0-preview/porting/porting-exporter-targets.html), I need to run `mbed export -i uvision --update-packs` to update whole `index.json`. But the related cmsis pack .pdsc `Nuvoton.NuMicro_DFP.pdsc` is not onlined yet. I have `Nuvoton.NuMicro_DFP.pdsc` locally, Is there a simple way to just updating `index.json` partially based on my local `Nuvoton.NuMicro_DFP.pdsc`?

#### Target(s) affected by this defect ?

N/A

#### Toolchain(s) (name and version) displaying this defect ?

N/A

#### What version of Mbed-os are you using (tag or sha) ?
7fce7f552c Merge pull request #12572 from rajkan01/waitapi_remove_deprecated

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

mbed-cli: 1.10.1

#### How is this defect reproduced ? 

N/A

ccli8, 2020-04-14:
> Please update the issue header with the missing information, the issue will not be mirroredto our internal defect tracking system or investigated until this has been fully resolved.

@ciarmcom Updated

0xc0170, 2020-04-14:
@ccli8 it should be possible - cmsis pack manager provides "downloaded only" packs. I can't see this however implemented in arm pack manager in tools.

cc @ARMmbed/mbed-os-tools Please review