ARMmbed/mbed-os #12803: NU_PFM_M2351_NPSA_S: failing nightly in ATHandler - not supporting serial

0xc0170, 2020-04-14:
### Description of defect

`NU_PFM_M2351_NPSA_S` fails with an error:

```
Compile [  9.3%]: ATHandler.cpp
[Error] ATHandler.cpp@1548,17: unknown type name 'BufferedSerial'
[Error] ATHandler.cpp@1548,49: no member named 'set_baud' in 'mbed::FileHandle'
[ERROR] ./features/cellular/framework/device/ATHandler.cpp:1548:17: error: unknown type name 'BufferedSerial'
    static_cast<BufferedSerial *>(_fileHandle)->set_baud(baud_rate);
                ^
./features/cellular/framework/device/ATHandler.cpp:1548:49: error: no member named 'set_baud' in 'mbed::FileHandle'
    static_cast<BufferedSerial *>(_fileHandle)->set_baud(baud_rate);
```

The target has this in targets.json file:
```
        "device_has_remove": [
            "SERIAL",
            "SERIAL_ASYNCH",
            "SERIAL_FC",
            "STDIO_MESSAGES",
            "LPTICKER"
        ],
```

It does not support SERIAL, therefore the object BufferedSerial does not exists. A question is how ATHandler should handle this?

@ARMmbed/mbed-os-wan  Please review

cc @ARMmbed/team-nuvoton

#### Target(s) affected by this defect ?

NU_PFM_M2351_NPSA_S

#### Toolchain(s) (name and version) displaying this defect ?

all

#### What version of Mbed-os are you using (tag or sha) ?

33563b3cfb

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

not relevant

#### How is this defect reproduced ? 

Build the target (test or an example) and should fail

ccli8, 2020-04-14:
This is re-raise of https://github.com/ARMmbed/mbed-os/pull/11594#issuecomment-544323169. Maybe follow the suggestion or exclude NU_PFM_M2351_NPSA_S from CI.

kivaisan, 2020-04-14:
From `ATHandler` point of view, `set_baud_rate` can be ignored if serial is not supported, so it is easy to flag out. On the other hand the whole AT based cellular can be disabled if serial is not supported, but that is a more tricky task.

PR #12804 (2020-04-15): Cellular: Check if serial is supported

### Summary of changes 

ATHandler only support changing baud rate via `BufferedSerial` object. This fixes the build issue when serial is not supported.

Fixes #12803 

#### Impact of changes 

#### Migration actions required
closed 2020-04-15 by 0xc0170