ARMmbed/mbed-os #12806: NUCLEO_F767ZI failed with features-storage-tests-kvstore-general_tests_phase_1&2

jamesbeyond, 2020-04-14:
### Description of defect
With the PR of https://github.com/ARMmbed/mbed-os/pull/12638, kvstore tests been enabled on a number of targets.

One of the issue been noticed is 
`features-storage-tests-kvstore-general_tests_phase_1&2`
is failing on NUCLEO_F767ZI with nightly tests:

logs:
http://mbed-os-ci.s3-website-eu-west-1.amazonaws.com/?prefix=jenkins-ci/ARMmbed/mbed-os/mbed-os-ci-nightly/artifacts/master/509/greentea-test/151/FAIL/NUCLEO_F767ZI/

#### Target(s) affected by this defect ?
NUCLEO_F767ZI

#### Toolchain(s) (name and version) displaying this defect ?
GCC_ARM and ARMC6

#### What version of Mbed-os are you using (tag or sha) ?

latest master or 098c72a31277e0b292dd6cdf1abaf9ca9d487032

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

#### How is this defect reproduced ? 
mbed test -t GCC_ARM -m NUCLEO_F767ZI -n features-storage-tests-kvstore-general_tests_phase_1

jamesbeyond, 2020-04-14:
@tymoteuszblochmobica @SeppoTakalo

jeromecoutant, 2020-04-15:
Issue:

````
[1586830838.28][CONN][RXD] >>> Running case #1: 'TDB_kvstore_init'...
[1586830839.38][CONN][RXD] ++ MbedOS Error Info ++
[1586830839.38][CONN][RXD] Error Status: 0x80FF0144 Code: 324 Module: 255
[1586830839.48][CONN][RXD] Error Message: Assertion failed: _area_params[0].size == _area_params[1].size
[1586830839.48][CONN][RXD] Location: 0x8014F8F
[1586830839.58][CONN][RXD] File: ./features/storage/kvstore/tdbstore/TDBStore.cpp+215
[1586830839.58][CONN][RXD] Error Value: 0x0
[1586830839.68][CONN][RXD] Current Thread: main Id: 0x20006094 Entry: 0x8015C9B StackSize: 0x1000 StackMem: 0x20004B00 SP: 0x200058B0
[1586830839.78][CONN][RXD] For more info, visit: https://mbed.com/s/error?error=0x80FF0144&tgt=NUCLEO_F767ZI
[1586830839.78][CONN][RXD] -- MbedOS Error Info --
````

0xc0170, 2020-04-15:
cc @ARMmbed/mbed-os-storage

jeromecoutant, 2020-04-29:
Any update from @ARMmbed/mbed-os-storage ?
Thx

0xc0170, 2020-05-06:
I've seen this in the recent nightlies, going to talk to the team about it

VeijoPesonen, 2020-05-07:
My first guess would be that the board is using non-uniform erase sector sizes. Other option is that there is so little space(required number of erase sectors for KVStore which is at minimum two) it doesn't fit.
```
    // The size calculations are a bit complex because we need to make sure we're
    // always aligned to an erase block boundary (whose size may not be uniform
    // across the address space), and we also need to make sure that the first
    // area never goes over half of the total size even if bd_size is an odd
    // number of sectors.
    while (true) {
        bd_size_t erase_unit_size = _bd->get_erase_size(area_0_size);
        if (area_0_size + erase_unit_size <= (bd_size / 2)) {
            area_0_size += erase_unit_size;
        } else {
            break;
        }
    }
    while (true) {
        bd_size_t erase_unit_size = _bd->get_erase_size(area_0_size + area_1_size);
        if (area_1_size + erase_unit_size <= (bd_size / 2)) {
            area_1_size += erase_unit_size;
        } else {
            break;
        }
    }
    _area_params[0].address = 0;
    _area_params[0].size = area_0_size;
    _area_params[1].address = area_0_size;
    _area_params[1].size = area_1_size;
    // The areas must be of same size
    MBED_ASSERT(_area_params[0].size == _area_params[1].size);
```

VeijoPesonen, 2020-05-07:
I'm not seeing the same issue if I try to test in RAAS with latest Mbed OS. Not saying that I don't see any issues as there seems to be some. I would say that with debug build the application takes so much space there is no more left enough for KVStore. Situation might vary depending how much application happens to take space.

In case of the Mbed OS version  029109a2f020fb54813e616e35a735df73b540b1 the issue doesn't become visible - even with debug build. If I switch to same version as has been mentioned in the Mbed OS issue ticket the problem becomes visible with debug- but not with release-build. But that doesn't tell that the KVStore regions are same size, assertions are just disabled.
```
[1588835387.27][CONN][INF] found KV pair in stream: {{__testcase_finish;FS_set_buffer_size_is_zero;1;0}}, queued...
[1588835387.37][CONN][RXD] >>> Running case #44: 'FS_set_same_key_several_time'...
[1588835387.47][CONN][INF] found KV pair in stream: {{__testcase_start;FS_set_same_key_several_time}}, queued...
[1588835390.49][CONN][RXD] lfs warn:313: No more free space 7
[1588835390.59][CONN][RXD] lfs warn:313: No more free space 7
[1588835391.59][CONN][RXD] :259::FAIL: Expected 0 Was 271
[1588835395.82][CONN][RXD] >>> 'FS_set_same_key_several_time': 0 passed, 1 failed with reason 'Test Cases Failed'
[1588835395.82][CONN][RXD]
[1588835395.82][CONN][INF] found KV pair in stream: {{__testcase_finish;FS_set_same_key_several_time;0;1}}, queued...
[1588835395.92][CONN][RXD] >>> Running case #45: 'FS_set_several_keys_multithreaded'...
[1588835396.02][CONN][INF] found KV pair in stream: {{__testcase_start;FS_set_several_keys_multithreaded}}, queued...
[1588835400.06][CONN][RXD] lfs warn:313: No more free space 7
[1588835400.06][CONN][RXD] lfs warn:313: No more free space 7
[1588835402.15][CONN][RXD] :272:lfs w:FAILarn:3: Ex13: Npecteo mord 0 We fras 27ee sp1
[1588835402.15][CONN][RXD] ace 7
[1588835402.15][CONN][RXD] lfs warn:313: No more free space 7
[1588835403.15][CONN][RXD] :272::FAIL: Expected 0 Was 271
[1588835403.26][CONN][RXD] :297::FAIL: Expected 0 Was 263
[1588835403.26][CONN][RXD] :297::FAIL: Expected 0 Was 263
[1588835404.27][CONN][RXD] :303::FAIL: Expected 0 Was 263
[1588835404.36][CONN][RXD] :303::FAIL: Expected 0 Was 263
[1588835404.36][CONN][INF] found KV pair in stream: {{__testcase_finish;FS_set_several_keys_multithreaded;0;6}}, queued...
[1588835404.47][CONN][RXD] >>> 'FS_set_several_keys_multithreaded': 0 passed, 6 failed with reason 'Test Cases Failed'
[1588835434.32][CONN][RXD] >>> Running case #53: 'FS_set_several_key_value_sizes'...
[1588835434.42][CONN][INF] found KV pair in stream: {{__testcase_start;FS_set_several_key_value_sizes}}, queued...
[1588835438.44][CONN][RXD] lfs warn:313: No more free space 7
[1588835438.44][CONN][RXD] lfs warn:313: No more free space 7
[1588835439.55][CONN][RXD] :462::FAIL: Expected 0 Was 271
[1588835440.55][CONN][RXD] lfs warn:313: No more free space 7
[1588835440.55][CONN][RXD] lfs warn:313: No more free space 7
[1588835441.68][CONN][RXD] :462::FAIL: Expected 0 Was 271
[1588835442.67][CONN][RXD] lfs warn:313: No more free space 7
[1588835442.67][CONN][RXD] lfs warn:313: No more free space 7
[1588835443.77][CONN][RXD] :462::FAIL: Expected 0 Was 271
[1588835444.78][CONN][RXD] lfs warn:313: No more free space 7
[1588835444.78][CONN][RXD] lfs warn:313: No more free space 7
[1588835445.88][CONN][RXD] :462::FAIL: Expected 0 Was 271
[1588835446.90][CONN][RXD] :469::FAIL: Expected 0 Was 263
[1588835446.99][CONN][RXD] :470::FAIL: Expected 'name_b' Was ''
[1588835446.99][CONN][INF] found KV pair in stream: {{__testcase_finish;FS_set_several_key_value_sizes;0;6}}, queued...
[1588835447.09][CONN][RXD] >>> 'FS_set_several_key_value_sizes': 0 passed, 6 failed with reason 'Test Cases Failed'
```

VeijoPesonen, 2020-05-07:
I would say there need alter the build system in a way that KVStore is treated in a similar manner as bootloader. In other words certain space(region) is reserved for it and error is reported if there is an overlap. I don't know is our build system that clever but I would hope so.

```
Using ROM regions bootloader, header, application, post_application in this build.
  Region bootloader: size 0x7000, offset 0x0
  Region header: size 0x70, offset 0x8000
  Region application: size 0x77c00, offset 0x8400
  Region post_application: size 0x80000, offset 0x80000

```

_post_application_ would be something what is reserved for KVStore. ~~The default KVStore configuration leaves nowadays two spare sectors(erase sectors) for the application to grow.~~ Scratch that, that was old behavior. Nowadays KVStore tries to take two sectors from the end of the flash.

If bootloader is used this issue can be avoided by placing the KVStore between the bootloader and the application.

jeromecoutant, 2020-06-08:
>  0xc0170 added the devices: st label on 15 Apr

Are you sure ?
Issue seems to be in features/storage/kvstore/tdbstore/TDBStore.cpp file

@ARMmbed/mbed-os-storage
@MarceloSalazar

LDong-Arm, 2020-08-17:
I can reproduce this issue 100% of the time no matter which toolchain, profile and revision I use.

LDong-Arm, 2020-08-20:
Fix created: https://github.com/ARMmbed/mbed-os/pull/13466

In short, the STM32F7 flash has only a few very large sectors of variable sizes, and we had alignment issues with FlashIAPBlockDevice and littlefs v1 lacks support when sectors are few.

LDong-Arm, 2020-09-10:
> Issue:
>
> ```
> [1586830838.28][CONN][RXD] >>> Running case #1: 'TDB_kvstore_init'...
> [1586830839.38][CONN][RXD] ++ MbedOS Error Info ++
> [1586830839.38][CONN][RXD] Error Status: 0x80FF0144 Code: 324 [...]

This is harder to reproduce, but also caused by non-uniform erase size. TDBStore requires two block of the same size and uses the first two blocks, whose sizes are not guaranteed.

LDong-Arm, 2020-09-14:
Just to make it clear, we agreed with ST to skip LittleFileSystem on STM32F2/4/7 (#13593) and add to our documentation how to enable and properly align that manually.

PR #13593 (2020-09-17): STM32F: skip LittleFileSystem default instance and TDBStore tests

### Summary of changes 

Fixes: #12806

STM32F2/4/7 internal flashes have non-uniform sector layouts – a few smaller sectors (16/32KB), followed by some large ones (e.g. up to 128/256KB) that form most of the total flash. This leads to alignment issues of LittleFileSystem which only supports one constant erase size, whereas we might get a mixture of smaller and larger sectors. This is largely reproducible on F7 targets, though F2/F4's small sectors have less combined capacity so only large ones get used by luck.

This PR skips  [... cut]
closed 2020-09-17 by 0xc0170