ARMmbed/mbed-os #12807: USB msd test is faiiling on python 3

maciejbocianski, 2020-04-15:
### Description of defect

Test msd (`tests-usb_device-msd`) is not working when run under python3
With python 2.7 works fine.
Tested on win10

python 2.7 results:
```
| target     | platform_name | test suite           | test case                                       | passed | failed | result | elapsed_time (sec) |
|------------|---------------|----------------------|-------------------------------------------------|--------|--------|--------|--------------------|
| K64F-ARMC6 | K64F          | tests-usb_device-msd | mount/unmount and data test - Heap block device | 1      | 0      | OK     | 19.31              |
| K64F-ARMC6 | K64F          | tests-usb_device-msd | mount/unmount test - Heap block device          | 1      | 0      | OK     | 50.55              |
| K64F-ARMC6 | K64F          | tests-usb_device-msd | storage initialization                          | 1      | 0      | OK     | 0.07               |

```

python 3 results:
```
| target     | platform_name | test suite           | test case                                       | passed | failed | result  | elapsed_time (sec) |
|------------|---------------|----------------------|-------------------------------------------------|--------|--------|---------|--------------------|
| K64F-ARMC6 | K64F          | tests-usb_device-msd | mount/unmount and data test - Heap block device | 0      | 0      | SKIPPED | 0.0                |
| K64F-ARMC6 | K64F          | tests-usb_device-msd | mount/unmount test - Heap block device          | 0      | 1      | FAIL    | 32.61              |
| K64F-ARMC6 | K64F          | tests-usb_device-msd | storage initialization                          | 1      | 0      | OK      | 0.07               |
```

When running with python 3 msd mounts on host but python host script fails to detect it.
Likely python libs used to read disks/partition info doesn't work well with python3 (see WMI and pywin32 libs) https://github.com/ARMmbed/mbed-os/blob/35c93011eabf7823cb133f3f9b2c4657f2cb33a9/TESTS/host_tests/pyusb_msd.py#L188-L196

#### Target(s) affected by this defect ?
all

#### Toolchain(s) (name and version) displaying this defect ?

#### What version of Mbed-os are you using (tag or sha) ?
latest master
64bc9d9dd7cdd0c6bc8d73511a976f35d694f5a0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
$ pip list
Package            Version
------------------ -------------
appdirs            1.4.3
asn1ate            0.6.0
beautifulsoup4     4.6.3
cbor               1.0.0
certifi            2020.4.5.1
cffi               1.14.0
chardet            3.0.4
Click              7.0
cmsis-pack-manager 0.2.10
colorama           0.3.9
cryptography       2.9
ecdsa              0.15
fasteners          0.15
future             0.16.0
hidapi             0.7.99.post21
icetea             1.2.4
idna               2.7
intelhex           2.2.1
Jinja2             2.10.3
jsonmerge          1.7.0
jsonschema         2.6.0
junit-xml          1.8
lockfile           0.12.2
manifest-tool      1.5.2
MarkupSafe         1.1.1
mbed-cloud-sdk     2.0.8
mbed-flasher       0.10.1
mbed-greentea      1.7.4
mbed-host-tests    1.5.10
mbed-ls            1.7.10
mbed-os-tools      0.0.12
milksnake          0.1.5
monotonic          1.5
pip                20.0.2
prettytable        0.7.2
protobuf           3.5.2.post1
psutil             5.6.6
pyasn1             0.2.3
pycparser          2.20
pycryptodome       3.9.7
pyelftools         0.25
pyparsing          2.4.7
pyserial           3.4
python-dateutil    2.8.1
python-dotenv      0.12.0
pyusb              1.0.2
pywin32            224
PyYAML             4.2b1
requests           2.20.1
semver             2.9.1
setuptools         46.1.3
six                1.12.0
urllib3            1.24.2
wheel              0.34.2
WMI                1.4.9
yattag             1.13.2

#### How is this defect reproduced ? 
run usb-msd test on any supported target - use python 3
mbed test -m k64f -t ARM -n tests-usb_device-msd -v

@jamesbeyond @fkjagodzinski @paul-szczepanek-arm

maciejbocianski, 2020-05-25:
ping @rajkan01

rajkan01, 2020-05-31:
Already this issue was fixed and merged with PR#12944. Please close this issue.

maciejbocianski, 2020-06-01:
fixed by #12944

closed 2020-06-01 by maciejbocianski