ARMmbed/mbed-os #12809: CY8CPROTO_062_4343W hardfault with Greentea tests

jamesbeyond, 2020-04-15:
### Description of defect
CY8CPROTO_062_4343W hardfault with Greentea tests, after https://github.com/ARMmbed/mbed-os/pull/12679

```
[1586966794.34][HTST][INF] starting host test process...
[1586966795.05][CONN][INF] starting connection process...
[1586966795.05][CONN][INF] notify event queue about extra 60 sec timeout for serial port pooling
[1586966795.05][CONN][INF] initializing serial port listener...
[1586966795.05][PLGN][INF] Waiting up to 60 sec for '190113018b0c131e028b0c1300000000000000002e127069' serial port (current is 'COM31')...
[1586966795.05][HTST][INF] setting timeout to: 60 sec
[1586966795.08][SERI][INF] serial(port=COM31, baudrate=9600, read_timeout=0.01, write_timeout=5)
[1586966795.08][SERI][INF] reset device using 'default' plugin...
[1586966795.33][SERI][INF] waiting 1.00 sec after reset
[1586966796.33][SERI][INF] wait for it...
[1586966796.33][SERI][TXD] mbedmbedmbedmbedmbedmbedmbedmbedmbedmbed
[1586966796.33][CONN][INF] sending up to 5 __sync packets (specified with --sync=5)
[1586966796.33][CONN][INF] sending preamble '517ec68b-8897-45b0-8161-b53b72a84f43'
[1586966796.33][SERI][TXD] {{__sync;517ec68b-8897-45b0-8161-b53b72a84f43}}
[1586966796.35][CONN][RXD]
[1586966796.35][CONN][RXD] ++ MbedOS Fault Handler ++
[1586966796.35][CONN][RXD]
[1586966796.35][CONN][RXD] FaultType: HardFault
[1586966796.35][CONN][RXD]
[1586966796.35][CONN][RXD] Context:
[1586966796.35][CONN][RXD] R0: 0
[1586966796.35][CONN][RXD] R1: E03
[1586966796.35][CONN][RXD] R2: C000001
[1586966796.35][CONN][RXD] R3: 80000E03
[1586966796.35][CONN][RXD] R4: 1
[1586966796.35][CONN][RXD] R5: 17
[1586966796.35][CONN][RXD] R6: 4026FF1C
[1586966796.35][CONN][RXD] R7: 100128AB
[1586966796.35][CONN][RXD] R8: 1249
[1586966796.35][CONN][RXD] R9: C000001
[1586966796.35][CONN][RXD] R10: 100128EC
[1586966796.35][CONN][RXD] R11: 100128EC
[1586966796.35][CONN][RXD] R12: 16001820
[1586966796.35][CONN][RXD] SP   : 80FFF98
[1586966796.35][CONN][RXD] LR   : 1000624B
[1586966796.35][CONN][RXD] PC   : 4022002C
[1586966796.35][CONN][RXD] xPSR : 60000000
[1586966796.35][CONN][RXD] PSP  : 7FACBFFC
[1586966796.35][CONN][RXD] MSP  : 80FFF78
[1586966796.35][CONN][RXD] CPUID: 410FC241
[1586966796.35][CONN][RXD] HFSR : 40000000
[1586966796.35][CONN][RXD] MMFSR: 1
[1586966796.35][CONN][RXD] BFSR : 0
[1586966796.35][CONN][RXD] UFSR : 0
[1586966796.35][CONN][RXD] DFSR : 0
[1586966796.35][CONN][RXD] AFSR : 0
[1586966796.35][CONN][RXD] Mode : Thread
[1586966796.35][CONN][RXD] Priv : Privileged
[1586966796.35][CONN][RXD] Stack: MSP
[1586966796.35][CONN][RXD]
[1586966796.35][CONN][RXD] -- MbedOS Fault Handler --
[1586966796.35][CONN][RXD]
[1586966796.35][CONN][RXD]
[1586966796.35][CONN][RXD]
[1586966796.35][CONN][RXD] ++ MbedOS Error Info ++
[1586966796.35][CONN][RXD] Error Status: 0x80FF013D Code: 317 Module: 255
[1586966796.35][CONN][RXD] Error Message: Fault exception
[1586966796.35][CONN][RXD] Location: 0x4022002C
[1586966796.35][CONN][RXD] Error Value: 0x8001EF0
[1586966796.35][CONN][RXD] Current Thread: <unnamed> Id: 0x0 Entry: 0x2A00400A StackSize: 0x18894923 StackMem: 0x4A27E003 SP: 0x80FFF98
[1586966796.35][CONN][RXD] For more info, visit: https://mbed.com/s/error?error=0x80FF013D&tgt=CY8CPROTO_062_4343W
[1586966796.35][CONN][RXD] -- MbedOS Error Info --
```

#### Target(s) affected by this defect ?

CY8CPROTO_062_4343W 
Could be more Cypress targets

#### Toolchain(s) (name and version) displaying this defect ?

GCC_ARM and ARMC6

#### What version of Mbed-os are you using (tag or sha) ?

f011ccb2cb714fb763980062b58b563b940a44fa -> passing
b5f00bc67adde82d2e5ffb9e81e4afea8365a1f3  -> failing 

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

latest mbed-cli

#### How is this defect reproduced ? 

run any greentea tests, e.g.:
`mbed test -m CY8CPROTO_062_4343W -t ARM -n *crc*`

jamesbeyond, 2020-04-15:
@ARMmbed/team-cypress

vmedcy, 2020-04-15:
Hi, please check the latest master - fix is already applied: https://github.com/ARMmbed/mbed-os/pull/12752

0xc0170, 2020-04-16:
I can confirm, looking at the latest nightly run. The test passed (only that sleep known issue there). I'll close this as fixed.

closed 2020-04-16 by 0xc0170