ARMmbed/mbed-os #12812: High current consuption

bbugl, 2020-04-16:
### Description of defect
When calling mbedtls_platform_setup and subsequently mbedtls_platform_teardown current consumption stays high. The idle hook is never called after that.

What I have been able to determine:
- The reference counter works. crypto_platform_setup and crypto_platform_terminate are called correctly.
- NRF_CRYPTOCELL->ENABLE is turned on and off correctly.
- SaSi_LibInit is called and returns 0 (= SUCCESS).

Ron Eldor from ARM support (@roneld01 on the mbed forums, and, I guess, @RonEld on github) suggested to me to create this issues. He speculated that

> the Cryptocell clock was initiated in mbedtls_platform_setup(), but not terminated in mbedtls_platform_teardown() like it shoulde have been.

#### Target(s) affected by this defect ?
MCU_NRF52840
We use this target as a root for our custom targets.

#### Toolchain(s) (name and version) displaying this defect ?
ARM GCC 6.3.1
ARM GCC 9.2.1

#### What version of Mbed-os are you using (tag or sha) ?
mbed-os-5.15.1
hash: e642a7d8b3609a7c903e042cd772f00a5d299088

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
mbed-cli 1.10.2

#### How is this defect reproduced ? 
Pretty easily.
I just call mbedtls_platform_setup and after that mbedtls_platform_teardown.
I tried to add delays in between to resolve any timing issues, but that didn't help at all.

0xc0170, 2020-04-16:
cc @ARMmbed/mbed-os-crypto

bbugl, 2020-11-24:
We recently upgraded to mbed-6.4.0 but there is no change in this. any news about this?

closed 2021-10-20 by ciarmcom