ARMmbed/mbed-os #12821: DeviceKey: Key size mismatch between generate_root_of_trust() and mbed-bootloader

ccli8, 2020-04-17:
### Description of defect

#### Target(s) affected by this defect ?

n/a

#### Toolchain(s) (name and version) displaying this defect ?

n/a

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-6.0.0-alpha-3

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

n/a

#### How is this defect reproduced ? 

`mbed-bootloader` requires 16 bytes device key (https://github.com/ARMmbed/mbed-bootloader/blob/94579e9f238b15f7fabe14005cdd181da0fa9123/source/kvstore_rot.cpp#L25). For convenience and development, `mbed-bootloader` enabled application can call `DeviceKey`'s `generate_root_of_trust()` with mbedtls entropy configuration having been there to install device key. According to its implementation, it installs 32 bytes device key, which mismatches `mbed-bootloader` and causes error (https://github.com/ARMmbed/mbed-bootloader/blob/94579e9f238b15f7fabe14005cdd181da0fa9123/source/kvstore_rot.cpp#L64).

Suggest refining `generate_root_of_trust()` or providing its variant which can specify device key size parameter.

0xc0170, 2020-04-17:
cc @ARMmbed/mbed-os-storage

SeppoTakalo, 2020-04-17:
Thanks..
This is a valid issue..
Bootloader releases are build against 5.15 branch, so this does not affect them, but for future, it must be fixed before 6.0 release.

SeppoTakalo, 2020-04-17:
Fixed in https://github.com/ARMmbed/mbed-os/pull/12823

ccli8, 2020-04-20:
> Could you add some more detail to the description? A good description should be at least 25 words.

@ciarmcom Updated.

PR #12823 (2020-04-30): Allow Devicekey::generate_root_of_trust() to define key size.

### Summary of changes 

Allow `Devicekey::generate_root_of_trust()` to define key size.

By default, generate 16 byte keys, to be compatible with bootloader.
But allow user to generate 32 byte keys as well.

This fixes #12821 

#### Impact of changes 

Default key, generated by `Devicekey::generate_root_of_trust()` is now 16 bytes,
instead of 32 bytes.

#### Migration actions required 

If you want to generate 32 byte keys with `Devicekey::generate_root_of_trust()`,
you need to explicitly
closed 2020-04-30 by 0xc0170