ARMmbed/mbed-os #12822: KVStore: Potential buffer overrun in TDBStore

ccli8, 2020-04-17:
### Description of defect

#### Target(s) affected by this defect ?

n/a

#### Toolchain(s) (name and version) displaying this defect ?

n/a

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-6.0.0-alpha-3

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

n/a

#### How is this defect reproduced ? 

`reserved_data_buf_size` limits size of `reserved_data` buffer.

https://github.com/ARMmbed/mbed-os/blob/7b0c38aabba581080a80d0265c408a76a2c5a943/features/storage/kvstore/tdbstore/TDBStore.cpp#L1392-L1397

`reserved_data_buf_size` is not used for guard and `memcpy(...)` can cause buffer overrun  when `trailer.data_size` is larger than `reserved_data_buf_size`.

https://github.com/ARMmbed/mbed-os/blob/7b0c38aabba581080a80d0265c408a76a2c5a943/features/storage/kvstore/tdbstore/TDBStore.cpp#L1423-L1428

0xc0170, 2020-04-17:
cc @ARMmbed/mbed-os-storage

ccli8, 2020-04-20:
> Could you add some more detail to the description? A good description should be at least 25 words.

@ciarmcom Updated.

0xc0170, 2020-04-27:
@ccli8 Can you send PR fixing this?

ccli8, 2020-04-28:
> Can you send PR fixing this?

Fixed in #12875

PR #12875 (2020-05-12): KVStore: Fix buffer overrun when device key size doesn't match

### Summary of changes 

This PR tries to fix #12822, where buffer may overrun when injected device key is 32-byte but read as 16-byte.

----------------------------------------------------------------------------------------------------------------
### Pull request type 

    [x] Patch update (Bug fix / Target update / Docs update / Test update / Refactor)
    [] Feature update (New feature / Functionality change / New API)
    [] Major update (Breaking change E.g. Return code change / API behaviour change)

--------------------- [... cut]
closed 2020-05-12 by 0xc0170