ARMmbed/mbed-os #12829: Which branch/tag should I use to import Mbed OS version 2 library in my project?

jadhavshital, 2020-04-20:
Hi,
I need to detect blur images for that I am using opencv(https://os.mbed.com/users/joeverbout/code/main/) library. This library is depends on mbed-os version 2 library. 

Which branch/tag should I use to import mbed-os 2 library?

Kindly suggest some solution.
Thanks in advance.

0xc0170, 2020-04-20:
Please use forum.mbed.com for questions. Thank you

To answer, https://os.mbed.com/users/mbed_official/code/mbed/ this is the one you are looking for

closed 2020-04-20 by 0xc0170

jadhavshital, 2020-04-21:
I have already checked https://os.mbed.com/users/mbed_official/code/mbed/ this link but It is not cloning to my local. When I go to that link it says "An error occurred during processing your request" 
[image]

Your help will be appreciated.

0xc0170, 2020-04-22:
See https://github.com/ARMmbed/mbed-cli/issues/968 (there is a fix opened, if you use py3), just fetched that fix and should work if not , report it on the forums. as it might be either website issue or mbed-cli