ARMmbed/mbed-os #12832: nightly/Prs: littlefs wear leveling tests failing

0xc0170, 2020-04-20:
### Description of defect

`features-storage-filesystem-littlefs-tests-filesystem_recovery-wear_leveling` tests are failing. PRs testing as of today and also the latest nightly

See http://mbed-os-ci.s3-website-eu-west-1.amazonaws.com/?prefix=jenkins-ci/ARMmbed/mbed-os/mbed-os-ci-nightly/artifacts/master/513/greentea-test/155/FAIL/NUCLEO_F429ZI/ARM/ or http://mbed-os-ci.s3-website-eu-west-1.amazonaws.com/?prefix=jenkins-ci/ARMmbed/mbed-os/mbed-os-ci-nightly/artifacts/master/513/greentea-test/155/FAIL/K64F/ARM/

These are from the nightly, also can see on Prs , here's one from today: https://github.com/ARMmbed/mbed-os/pull/12810#issuecomment-616530057

#### Target(s) affected by this defect ?

K64F and Nucleo F429

#### Toolchain(s) (name and version) displaying this defect ?

ARM and GCC ARM

#### What version of Mbed-os are you using (tag or sha) ?

27405bff31d4c31f24e7e07f2b7ad99c4716bd47

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

Not relevant

#### How is this defect reproduced ? 

Rerun littlefs tests

cc @ARMmbed/mbed-os-storage @ARMmbed/mbed-os-test

0xc0170, 2020-04-20:
As this just appears today (and only in the latest nightly, I can't find the failures in the nightly from 2 days ago), neither PRs on Friday, is this target failure related?

0xc0170, 2020-04-20:
Looking at what has changed, nothing since Friday so should not be a code change, but rather something in targets in CI

0xc0170, 2020-04-23:
The latest one can be found :http://mbed-os-ci.s3-website-eu-west-1.amazonaws.com/?prefix=jenkins-ci/ARMmbed/mbed-os/mbed-os-ci/PR-12783/artifacts/12783/5/greentea-test/4109/FAIL/ (affecting more devices).

0xc0170, 2020-04-24:
Filesystem example tests have been failing in the recent 2 days, we suspect this could be related (http://mbed-os-ci.s3-website-eu-west-1.amazonaws.com/?prefix=jenkins-ci/ARMmbed/mbed-os/mbed-os-ci-lts-pr/PR-12862/artifacts/12862/1/example-test-lts/81/FAIL/DISCO_L475VG_IOT01A/ARM/)

VeijoPesonen, 2020-05-04:
The test failure surfaced after introduction of the commit 10481f2f7e460cd50f99d5a88c00a742bbfd3929. Changes made to ExhaustibleBlockDevice seem to trigger the issue.

VeijoPesonen, 2020-05-06:
Fixed by #12919.

closed 2020-05-06 by ciarmcom