ARMmbed/mbed-os #12837: Unable to link mbed-os aginst multiple applications with custom configuration

nat45928, 2020-04-20:
### Description of defect

I am unable to build a library from `mbed-os` to link against when using a custom `mbed_app.json` because the `mbed_config.h` produced with the `--library` option differs from a normally built application. This causes the application to fail to boot.

#### Additional Background

I have a project which produces multiple binary applications from a single codebase. I have been using the compiler `--source` option to change which files are selected by mbed-os. For example (where folders `foo` and `bar` contain different `main.cpp` files):

1. `mbed compile --app-config mbed_app.json --source foo --source mbed-os` builds `foo`
2. `mbed compile --app-config mbed_app.json --source bar --source mbed-os` builds `bar`

However, this takes a long time due to the rebuild of `mbed-os` library for each application. I would like to use the provided `--library` option to build `mbed-os` once then link my two applications aginst it. Similar to:

1. `mbed compile --app-config mbed_app.json --source mbed-os --build BUILD/mbed-os --library --no-archive` builds the `mbed-os` library
2. `mbed compile --app-config mbed_app.json --source foo --source BUILD/mbed-os --build BUILD/foo` builds `foo`
3. `mbed compile --app-config mbed_app.json --source bar --source BUILD/mbed-os --build BUILD/bar` builds `bar`

This process works but produces binaries that do not execute. Upon digging the resulting `mbed_config.h` for step 1 vs steps 2 and 3 differ. Digging into [the tools folder](tools/build.py) it appears that `build.py` does not respect the `app-config` CLI parameter (or `profile`).

Is this workflow expected to work, is there a better workflow that should be followed?

#### Target(s) affected by this defect ?

NUCLEO_F767ZI

#### Toolchain(s) (name and version) displaying this defect ?

GCC_ARM, 9-2019-q4-major, 9.2.1 20191025

#### What version of Mbed-os are you using (tag or sha) ?

d5f034b637957f9123e2a83257439eb3e1d80ca7

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

mbed-cli, 1.10.2

#### How is this defect reproduced ? 

1. Create a fresh project.
2. Create / edit the `mbed_app.json` configuration to change a parameter (`platform.stdio-baud-rate` for example).
3. Build `mbed-os` using `mbed compile --app-config mbed_app.json --source mbed-os --build BUILD/mbed-os --library --no-archive`
4. Build app1 using `mbed compile --app-config mbed_app.json --source foo --source BUILD/mbed-os --build BUILD/foo`
5. The application does not execute correctly.

0xc0170, 2020-04-22:
cc @ARMmbed/mbed-os-tools

nat45928, 2020-04-23:
It seems like there are a few changes needed:

1. mbed-cli's `mbed.py` needs to pass `--app_config` to `build.py` [here](https://github.com/ARMmbed/mbed-cli/blob/master/mbed/mbed.py#L2757).
2. `build.py` needs to accept `--app_config` by calling `get_default_options_parser(add_app_config=True)` [here](https://github.com/ARMmbed/mbed-os/blob/master/tools/build.py#L54).
3. `build.py` needs to pass `app_config` to the `build_library()` methods [here](https://github.com/ARMmbed/mbed-os/blob/master/tools/build.py#L191).

Making these changes locally allows me to build working applications. Is there anything wrong with this approach?

0xc0170, 2020-04-27:
Thanks @nat45928 for the details how to fix. We will review

nat45928, 2020-04-28:
@0xc0170 Is the approach I proposed appropriate? If so I am happy to open PRs to do the work. I would really like this workflow to be functional.

0xc0170, 2021-01-26:
@nat45928 starting the previous year we started working on moving to CMake based build. I'll close this and we should get a new issue. This should be feasible with the new tools (not using --library command but rather using cmake STATIC or OBJECT libraries).

closed 2021-01-26 by 0xc0170

nat45928, 2021-02-01:
Thanks for the heads up @0xc0170, I had some problems getting this setup so I opened a new issue (#14220) for the cmake based build system.