ARMmbed/mbed-os #12848: Issue with python packages install

jeromecoutant, 2020-04-22:
### Description of defect

New MBED-OS environment install:

I would like to install all python packages from mbed requirement.txt file,
but I got issue with hidapi package

````
$ pip install -r requirements.txt
...
Collecting hidapi<0.8.0,>=0.7.99
  Using cached hidapi-0.7.99.post21.tar.gz (51 kB)
    ERROR: Command errored out with exit status 1:
...
    Download error on https://pypi.org/simple/Cython/: [WinError 10061] No connection could be made because the target machine actively refused it -- Some packages may not be found!
    Couldn't find index page for 'Cython' (maybe misspelled?)
...
````

=> manual install of Cython
````
$ pip install Cython
Successfully installed Cython-0.29.16
````

=> new try:
````
$ pip install -r requirements.txt
...
ERROR: Could not find a version that satisfies the requirement pywin32==224 (from -r requirements.txt (line 26)) (from versions: 225, 226, 227)
...
````

=> I remove "==224" condition in requirements.txt file
````
$ pip install -r requirements.txt
...
    Running setup.py install for hidapi: started
    Running setup.py install for hidapi: finished with status 'error'
...
    error: Microsoft Visual C++ 14.0 is required. Get it with "Microsoft Visual C++ Build Tools": https://visualstudio.microsoft.com/downloads/
...
````

=> I then remove hidapi line from requirements.txt file...

OK

Last question is why mbed-cli is not part of requirements.txt ?

#### Target(s) affected by this defect ?

NA

#### Toolchain(s) (name and version) displaying this defect ?

NA

#### What version of Mbed-os are you using (tag or sha) ?

Master branch : 'mbed-os-99.99.99

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

$ python --version
Python 3.8.2

#### How is this defect reproduced ? 

$ pip install -r requirements.txt

0xc0170, 2020-04-23:
cc @ARMmbed/mbed-os-tools

madchutney, 2020-04-23:
Hi @jeromecoutant, we have seen some issues with the most recent version of `pip` and the manifest package which could be related. Could you please do a `pip list` so we can see the versions of these packages as well as letting us know the version of Mbed OS you are using?

Thanks

0xc0170, 2020-04-23:
A new env on windows (latest pip installed, python 3.6.8) , I was able to reproduce the second error: `pywin32==224` fails. 
If I update the version to 227 (the latest, as 224 was released Sept 2018, I assume they released it for older python versions thus failing), all is fine and installed.

madchutney, 2020-04-23:
It would seem sensible to me, but I don't know why the version was fixed in the first place. It will certainly mean that it will not be possible to build older versions of Mbed OS in the same environment if we do update it.

jeromecoutant, 2020-04-23:
About my Python version (3.8), is it OK ?
(as i can see in IC that  you are testing only 3.5/3.6/3.7)

madchutney, 2020-04-23:
It should be ok, but it hasn't been tested as you say. So there may be unknown issues.

jeromecoutant, 2020-04-24:
Hi @madchutney and @0xc0170 

As I am at home, I could try with my personal PC (no ST security or proxy stuff...) :-)

So from description, there are 4 issues:

- Cython install: OK, no issue
- pywin32: real issue
- hidapi needs Microsoft Visual C++ 14.0: real issue
- mbed-cli is missing: real issue

Note I uninstalled python 3.8, and installed python 3.7, hidapi issue disappear!

0xc0170, 2020-04-27:
Similar issue was reported earlier: https://github.com/ARMmbed/mbed-os/issues/12622 . Python 3.7 recommended for now.

jeromecoutant, 2020-06-08:
https://github.com/ARMmbed/mbed-os/blob/master/requirements.txt#L26

Issue still there...

petertorelli, 2020-09-25:
@jeromecoutant Still there ... are we just manually editing requirements.txt as a workaround?

jeromecoutant, 2020-09-28:
yes...

pip install pywin32

@MarceloSalazar

closed 2021-01-14 by jeromecoutant