ARMmbed/mbed-os #12857: TCP socket error get directly passed into mbedtls which displays a random error when traces are enabled.

ithinuel, 2020-04-23:
### Description of defect

When using a TLSSocket, the error code returned at the TCP layer and under are directly passed to mbedtls without convertion from `NSAPI_ERROR_*` to `MBEDTLS_ERR_*`.
When traces are enabled this results in `TLSSocketWrapper::print_mbedtls_error` passing an NSAPI error into `mbedtls_strerror` which reports some unrelated error message.
EG:
Unexpected socket closure, `TCPSocket::send` returns `-3004` (`NSAPI_ERROR_NO_CONNECTION`) which gets reported as :
`[ERR ][TLSW]: mbedtls_ssl_write() failed: -0x0bbc (-3004): UNKNOWN ERROR CODE (0B80) : ENTROPY - Critical entropy source failure`

This is extremely missleading.

#### Target(s) affected by this defect ?

All targets are affected.

#### Toolchain(s) (name and version) displaying this defect ?

#### What version of Mbed-os are you using (tag or sha) ?

5e765487c4f34ee1f348a598380d0fdd056b4420

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

```
mbed-cli           1.10.2
mbed-cloud-sdk     2.0.8
mbed-flasher       0.10.1
mbed-greentea      1.7.4
mbed-host-tests    1.5.10
mbed-ls            1.7.10
mbed-os-tools      0.0.12
```

#### How is this defect reproduced ? 

Try to send data on a TLSSocket closed remotely.

0xc0170, 2020-04-24:
cc @ARMmbed/mbed-os-ipcore

kivaisan, 2020-04-27:
Netsocket errors do not match with mbedtls error codes very well and those are in totally different range for this purpose. The netsocket error code code goes through mbedtls all the way to application. So I don't see this as a problem. Maybe a cosmetic issue, but doing conversion here makes things even more complicated imho.

ithinuel, 2021-02-18:
@kivaisan As mentioned in the description, the issue is that 
`TLSSocketWrapper::print_mbedtls_error` that's suppose to help us diagnose issues by translating obscure error code into human readable message (relying on `mbedtls_strerror`) indicates something complitly wrong ang miss leading !
This is extremely poor user experience.

> those are in totally different range for this purpose

Maybe a solution would be to detect error codes that are out of `MBEDTLS_ERR_` range and rely on some other translation or even a default message for “Network layer errors”.
Detecting the range and having an `nsapi_strerror`  method would do the trick in my opinion.

closed 2021-10-20 by ciarmcom

mdkf, 2022-01-24:
Confused me today when the socket closed and I got a Entropy error. I think this is worth fixing.