ARMmbed/mbed-os #12858: NUCLEO-L476RG never goes into deep sleep

rieonke, 2020-04-23:
### Description of defect

Hello. I'm following the [Power optimization tutorial](https://os.mbed.com/docs/mbed-os/v5.15/tutorials/power-optimization.html) using NUCLEO-L476RG board,
but it never goes into deep sleep mode.

I tried many ways, but all failed. What am I doing wrong? 
Thanks!

#### Target(s) affected by this defect ?
NUCLEO_L476RG

#### Toolchain(s) (name and version) displaying this defect ?
GCC_ARM arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 9-2019-q4-major) 9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-5.15.2

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
mbed-cli 1.10.2
#### How is this defect reproduced ? 
main.cpp file:
```c++

#include "mbed.h"

#define SAMPLE_TIME 10000// msec

int main() {

  uint64_t prev_idle_time = 0;

  while (true) {

    mbed_stats_cpu_t stats;
    mbed_stats_cpu_get(&stats);

    uint64_t diff = (stats.idle_time - prev_idle_time);
    double idle = (diff * 100.0) / (SAMPLE_TIME * 1000.0);// usec;
    double usage = 100 - idle;                            // usec;;
    prev_idle_time = stats.idle_time;

    printf("Idle: %.2f Usage: %.2f \n", idle, usage);
    printf("Uptime: %llu ", stats.uptime / 1000);
    printf("Sleep time: %llu ", stats.sleep_time / 1000);
    printf("Deep Sleep: %llu\n", stats.deep_sleep_time / 1000);

    thread_sleep_for(SAMPLE_TIME);
  }
}

```
mbed_app.json file:
```json
{
  "target_overrides": {
    "NUCLEO_L476RG": {
      "platform.stdio-baud-rate": 9600,
      "platform.stdio-buffered-serial": false,
      "platform.cpu-stats-enabled": 1,
      "platform.stdio-convert-newlines": true,
      "target.macros": []
    }
  }
}
```

and the output: 
```
Idle: 100.00 Usage: 0.00 
Uptime: 90665 Sleep time: 89995 Deep Sleep: 0
Idle: 100.00 Usage: 0.00 
Uptime: 100740 Sleep time: 99995 Deep Sleep: 0
Idle: 100.00 Usage: 0.00 
Uptime: 110816 Sleep time: 109995 Deep Sleep: 0

```

after add the "MBED_SLEEP_TRACING_ENABLED" macro:
```
Sleep locks held:
[id: mbed_rtx_idle., count: 1]
UNLOCK: mbed_rtx_idle.cpp, ln: 150, lock count: 0
LOCK: lp_ticker.c, ln: 462, lock count: 1
UNLOCK: lp_ticker.c, ln: 332, lock count: 0
LOCK: mbed_rtx_idle.cpp, ln: 148, lock count: 1
```

jeromecoutant, 2020-04-24:
Hi @rieonke 

I don't have a NUCLEO_L476RG with me, but I have a DISCO_L475VG_IOT01A with me, which is quite similar.
And I am sorry, but I don't currently reproduce issue...

0xc0170, 2020-04-24:
The target has `NUCLEO_L476RG` but from the sleep logs:

`LOCK: mbed_rtx_idle.cpp, ln: 148, lock count: 1` - this line 148 is in non tickless idle hook, isn't it? This would mean you are getting just shallow sleep.

jeromecoutant, 2020-04-24:
Oh, you are compiling with debug profile ?
(deep sleep is then disabled)

rieonke, 2020-04-24:
@jeromecoutant @0xc0170 THANKS!
You just saved my day !

I have tried the develop/release profile, but the results is the same.

I released that I'm using OpenOCD to download the binary. when switching to the mbed-cli, It works fine!

closed 2020-04-24 by rieonke