ARMmbed/mbed-os #12861: LWiP Fatal Run-time error when LWiP debug is enabled

teetak01, 2020-04-24:
### Description of defect

Enabling debugs for LwIP caused crash.

```
Network initialized, registering...

++ MbedOS Error Info ++
Error Status: 0x80FF0100 Code: 256 Module: 255
Error Message: Fatal Run-time error
Location: 0x2ED6D
Error Value: 0x0
Current Thread: lwip_tcpip Id: 0x20003D24 Entry: 0x30631 StackSize: 0x960 StackMem: 0x1FFF11C8 SP: 0x1FFF19C0 
For more info, visit: https://mbed.com/s/error?error=0x80FF0100&tgt=K64F
-- MbedOS Error Info --
p != NULL:731 in file ./mbed-os/features/lwipstack/lwip/src/core/lwip_pbuf.c
```

#### Target(s) affected by this defect ?

K64F

#### Toolchain(s) (name and version) displaying this defect ?

GCC_ARM

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-5.15.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

mbed-cli 1.10.2

#### How is this defect reproduced ? 

compile https://github.com/armmbed/mbed-cloud-client-example with:
```
+            "lwip.debug-enabled"                         :1,
```
in `mbed_app.json`

Probably reproducable with any application.

0xc0170, 2020-04-24:
cc @ARMmbed/mbed-os-ipcore

kivaisan, 2020-04-27:
Kevin had already fixed this in master, so I created a PR for 5.15-branch.

AnttiKauppila, 2020-04-27:
@teetak01 Ready to be verified

teetak01, 2020-04-27:
Thanks. Verified.

PR #12869 (2020-04-27): LWIP: don't pbuf_free(NULL) on socket close (5.15-branch)

### Summary of changes 

This is cherry-picking this fix from master to 5.15 -branch:

`pbuf_free(NULL)` causes an `LWIP_ASSERT`. Make the call conditional.

Fixes #12861

#### Impact of changes 

#### Migration actions required
closed 2020-04-27 by teetak01