ARMmbed/mbed-os #12870: CYESKIT-064B0S2-4343W Test Fails

maclobdell, 2020-04-27:
### Description of defect

Mbed OS Test fails (greentea) for new target

See attached log
[CYESKIT-064B0S2-4343W-gcc-test-results.txt](https://github.com/ARMmbed/mbed-os/files/4539598/CYESKIT-064B0S2-4343W-gcc-test-results.txt)

#### Target(s) affected by this defect ?

CYESKIT_064B0S2_4343W 

#### Toolchain(s) (name and version) displaying this defect ?

Tested with GCC for Arm 9.2.1

#### What version of Mbed-os are you using (tag or sha) ?

master branch from 21 April
sha: 31e3305c82d12e2758d3db49e2a5c8a59d18445b
(a few commits after PR https://github.com/ARMmbed/mbed-os/pull/12787 was merged)

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
mbed-cli           1.10.1
mbed-greentea      1.7.4
mbed-host-tests    1.5.10
mbed-os-tools      0.0.12

#### How is this defect reproduced ? 

I am testing PR 12787 (after it was merged) with a CYESKIT_064B0S2_4343W board that I provisioned with the attached policy. I attempted to set the internal flash as the Boot area and the SMIF as the Upgrade storage area.

I enabled the QSPI and QSPIF in the target configuration. Then added a mbed_app.json that defines ssid and password.
          "nsapi.default-wifi-security"               : "WPA_WPA2",
          "nsapi.default-wifi-ssid"                   : "\"myssid\"",
          "nsapi.default-wifi-password"               : "\"mypassword\""

Then ran integration tests  with this command: 

 >  mbed test -m CYESKIT_064B0S2_4343W -t GCC_ARM -n tests-integration-* -DINTEGRATION_TESTS

All the storage tests work indicating that the Quad SPI Flash and block device driver are working.  That is great!

However, the network tests fail.  It seems to not be able to boot up the app.It gets stuck.

Also, I tried Mbed OS general tests with this command and saw some failures. 

>mbed test -m CYESKIT_064S2_4343W -t GCC_ARM

0xc0170, 2020-04-27:
cc @ARMmbed/team-cypress

kyle-cypress, 2020-04-27:
@maclobdell , can you elaborate on the changes that you made to the target configuration to enable QSPI? In particular, did you change the value of CY_STORAGE_WIFI_DATA?

maclobdell, 2020-04-28:
@kyle-cypress - I deleted the lines with "QSPIF" and "QSPI" from "components_remove" and "device-has_remove" in targets.json in the section for this target. I did not add or change CY_STORAGE_WIFI_DATA.

        "components_remove": [
            "QSPIF"
        ],

        "device_has_remove": [
            "ANALOGOUT",
            "QSPI"
        ],

0xc0170, 2020-05-07:
@maclobdell as there was an issue on master, there was a fix for timeout just today. Can you retest and report back (we still have drifting issue (https://github.com/ARMmbed/mbed-os/issues/12940). I assume that tests like lp ticker still fails as of now.

maclobdell, 2020-05-12:
I just checked out the master branch. I did not make any modifications to targets.json. I am still seeing network test failures. However, it could be a problem with my hardware or a problem with how I provisioned the board. 

```
> mbed test -m CYESKIT_064B0S2_4343W -t GCC_ARM -n tests-netsocket-udp* -vv

[1589292663.33][CONN][INF] found SYNC in stream: {{__sync;03c6d2d4-fafd-4ece-a3f0-7e9ccfe263e3}} it is #0 sent, queued...
[1589292663.34][CONN][INF] found KV pair in stream: {{__version;1.3.0}}, queued...
[1589292663.34][CONN][INF] found KV pair in stream: {{__timeout;1200}}, queued...
[1589292663.34][CONN][INF] found KV pair in stream: {{__host_test_name;default_auto}}, queued...
[1589292663.34][HTST][INF] sync KV found, uuid=03c6d2d4-fafd-4ece-a3f0-7e9ccfe263e3, timestamp=1589292663.334568
[1589292663.34][HTST][INF] DUT greentea-client version: 1.3.0
[1589292663.34][HTST][INF] setting timeout to: 1200 sec
[1589292663.34][HTST][INF] host test class: '<class 'mbed_os_tools.test.host_tests.default_auto.DefaultAuto'>'
[1589292663.34][HTST][INF] host test setup() call...
[1589292663.34][HTST][INF] CALLBACKs updated
[1589292663.34][HTST][INF] host test detected: default_auto
[1589292664.08][CONN][RXD] Timeout while setting up the backplane, whd_bus_sdio_init failed at 307
mbedgt: :66::FAIL: Expected 0 Was -3012
[1589292664.08][CONN][RXD] :66::FAIL: Expected 0 Was -3012
[1589292664.09][CONN][RXD] >>> failure with reason 'Assertion Failed' during 'Test Setup Handler'
```

maclobdell, 2020-05-12:
FYI - I'm using this policy file [policy_single_stage_CM4_2m_smif.json.txt](https://github.com/ARMmbed/mbed-os/files/4616401/policy_single_stage_CM4_2m_smif.json.txt)

and this mbed_app.json contents 

```
{
    "target_overrides": {
        "*": {
          "platform.stdio-baud-rate"                  : 115200,
          "platform.stdio-convert-newlines"           : true,
          "platform.stdio-buffered-serial"            : true,
          "platform.stdio-flush-at-exit"              : true,
          "mbed-trace.enable"                         : true,
          "nsapi.default-wifi-security"               : "WPA_WPA2",
          "nsapi.default-wifi-ssid"                   : "\"SSID\"",
          "nsapi.default-wifi-password"               : "\"PASSWORD\""
        }
    }
}
```
and I checked out 8a254a9cc6a0b3377d14cdd264e381133f7c54b1

maclobdell, 2020-05-15:
After review with Cypress engineers, we believe the network failures could be due to the fact that I provisioned the boards to enable Quad SPI Flash, which is not supported by the version of the Cypress bootloader that was used in provisioning. I'm going to provision a new board and run tests again today to verify.

maclobdell, 2020-05-15:
I can confirm that after provisioning a new board w/o quad spi flash support, the network tests listed above are all passing.

closed 2020-05-18 by ciarmcom