ARMmbed/mbed-os #12879: NRF52840: Setting transmit power using Nordic API, doesn't seem to take effect on Cordio BLE Platform.

yogeshk19, 2020-04-28:
### Description of defect
Since there is no existing API in the Cordio platform to set the Transmit power, I am using the Nordic's API to set the radio transmit power, and based on our testing so far it doesn't seem to take effect. When the application starts up, I am setting the following in main.cpp.

```
//set the TX power to 8 dbm
NRF_RADIO->TXPOWER = (RADIO_TXPOWER_TXPOWER_Pos8dBm << RADIO_TXPOWER_TXPOWER_Pos);

//enable dc2dc
NRF_POWER->DCDCEN = 1;
```

It seems like the transmit power stays at the default transmit power, I am guessing at 0 dbm.

#### Target(s) affected by this defect ?
NRF52840_DK

#### Toolchain(s) (name and version) displaying this defect ?
GCC

#### What version of Mbed-os are you using (tag or sha) ?
MBED-OS-5.12.2

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
mbed-cli

#### How is this defect reproduced ? 
Add the above code to set the transmit power using the Nordic API, in any of the MBED-OS BLE Examples and run it on a NR52840 DK board and doesn't seem to take effect. Please use the BLE Cordio platform and not the softdevice.

Thanks,
Yogesh

0xc0170, 2020-04-29:
cc @ARMmbed/mbed-os-pan

yogeshk19, 2020-04-30:
any thoughts? This is affecting the BLE range as we are not able to apply a higher transmit power. Also just FYI, applying a lower data rate using the BLE 5 feature is not an option. 

Thanks,
Yogesh

paul-szczepanek-arm, 2020-05-01:
The tx power might be set by the stack itself later during radio init or even before transmission.

yogeshk19, 2020-05-01:
@paul-szczepanek-arm , where would you suggest then we set the tx power? I could try in the BLE stack initialization event handler that is invoked? But if you think that even before transmission the tx power is being modified, that would prevent me from applying the right transmit power. Can you please confirm when is the tx power setting applied in the cordio platform?

Thanks,
Yogesh

yogeshk19, 2020-05-13:
@paul-szczepanek-arm , @pan- any thoughts on this issue?

yogeshk19, 2020-06-01:
@pan-  Any thoughts?

paul-szczepanek-arm, 2020-06-18:
There is now tx power control on the nrf cordio:
https://github.com/ARMmbed/mbed-os/pull/13040

paul-szczepanek-arm, 2020-07-22:
tracked internally https://jira.arm.com/browse/IOTPAN-640

AGlass0fMilk, 2021-02-15:
@pan- @paul-szczepanek-arm Has this deficiency in the BLE API been addressed?

I don't recall a way to set the TX power and this leads me to believe it is not possible with Nordic targets still.

AGlass0fMilk, 2021-02-16:
@yogeshk19 As a workaround for now, I have been able to persistently change the TX power set by Cordio by adding a line (hack) to the Cordio sources here:

https://github.com/ARMmbed/mbed-os/blob/26606218ad9d1ee1c8781aa73774fd7ea3a7658e/connectivity/drivers/ble/FEATURE_BLE/TARGET_NORDIC/TARGET_NORDIC_CORDIO/TARGET_NRF5x/stack/sources/pal_bb_ble.c#L1036-L1076

I simply set the TX power and override the results of the operation by adding:

```
pChan->txPower = 8; // Whatever power setting you want here. 8 = +8dBm (maximum)
```
below line 1076.

@pan- @paul-szczepanek-arm I wasn't able to trace the extremely confusing Cordio WSF and scheduling system back to where the TX power parameter is set during the call to the above code. It doesn't seem like it would be terribly difficult to add this functionality to the Nordic Cordio implementation...

Point me to it and maybe I'll make a PR ;)

AGlass0fMilk, 2021-02-16:
See this:

https://github.com/AGlass0fMilk/mbed-os/commit/cbb3fd7127f60decc30a71541195925bad0b4b48

Spinnaker-design, 2022-03-02:
This does not appear to be fixed, it was just closed?