ARMmbed/mbed-os #12891: STM32WB55 RTC does not retain time

eriknayan, 2020-04-29:
### Description of defect

The RTC resets time/date when the main power supply is powered off, even though there's a supercap on the VBAT pin as a standby voltage supply (I've checked with a multi-meter that voltage is always present on VBAT). It appears that the `RTC_ISR_INITS` flag is being reset when the device is powered back on, because the `rtc_isenabled()` function always returns 0 unless a new `set_time()` call is done. It's a strange behavior since this could only happen if both supplies have previously been powered off (as described in `stm32wbxx_hal_rtc.c`).

#### Target(s) affected by this defect ?

STM32WB55RG 

#### Toolchain(s) (name and version) displaying this defect ?

GCC for Arm (gcc-arm-none-eabi-9-2019-q4-major)

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-5.15.2 - https://github.com/ARMmbed/mbed-os/commit/565ab149819481224ab43f878c3921b14b11d180

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

mbed-cli 1.10.2

#### How is this defect reproduced ? 

This bug can be reproduced following the RTC Time example (https://os.mbed.com/docs/mbed-os/v5.15/apis/rtc.html) and removing the `set_time(1256729737)` statement (because we want time to keep going). I also added a call to `rtc_isenabled()` in the main loop in order to verify that RTC was working.

0xc0170, 2020-04-30:
cc @ARMmbed/team-st-mcd

eriknayan, 2020-05-15:
Hello @ARMmbed/team-st-mcd @ciarmcom , any comments regarding this issue?

jeromecoutant, 2020-05-19:
Not a trivial question...
Seems that backup domain is reset when the main power supply is powered off.

Maybe you could read again paragraph 6.1.4 Battery backup domain
from MCU reference Manual...

jeromecoutant, 2020-05-20:
Hi
As a test, could you add this commit https://github.com/jeromecoutant/mbed/commit/97891405c04d560ce808ac9c0737fad8bd46b5d8 in your application ?

eriknayan, 2020-05-20:
Hi @jeromecoutant , thanks for replying

I've added the commit you mentioned in the project, but still the RTC does not retain the time.

After reading again the documentation and searching the web, I created a very simple project without mbed-OS using STMCubeIDE, and I've set the RTC Clock to LSE.

In this configuration, my custom board holds the calendar time, even after a power off.

Therefore I think there's something wrong with the clock selection in mbed-OS. I'll try to fix it and let you know here the results.

eriknayan, 2020-05-26:
Hello @jeromecoutant  

After verifying a lot of things, I think I figured out why the RTC for WB55RG kept resetting after a power off. In fact, in `system_clock.c`, inside `Configure_RF_Clock_Sources()`, the Backup Domain was being reset in lines 70-73:

```
HAL_PWR_EnableBkUpAccess();
HAL_PWR_EnableBkUpAccess();
__HAL_RCC_BACKUPRESET_FORCE();
__HAL_RCC_BACKUPRESET_RELEASE();
```

So I commented this lines and now the RTC (with LSE as clock source) keeps running under Vbat and recovers its value and keeps counting after power off/power on.

I don't know what's the impact of this on the RF configuration, but I'll run some tests here.

Do you think I can create a PR for this change?

jeromecoutant, 2020-05-26:
Hi

You can, of course, create a PR for any patch which correct some wrong behavior !

In that case, you have to create it on top mbed-os-5.15 branch,
as in master branch (mbed-os-6), it is already removed since #12384

eriknayan, 2020-05-26:
Ok! I will create the PR. Thanks for your help !

closed 2020-05-26 by eriknayan