ARMmbed/mbed-os #12907: get_i2c_timing function uses wrong SysClock value

mrdernst, 2020-04-30:
### Description of defect

When underclocking the MCU, the I2C peripherals are not working anymore because the wrong system clock is used/checked for setting the right I2C timing.
#### Target(s) affected by this defect ?

STM32

#### Toolchain(s) (name and version) displaying this defect ?

toolchain-gccarmnoneeabi 1.70201.0 (7.2.1)

#### What version of Mbed-os are you using (tag or sha) ?

5.14.1

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

VSCode Version 1.44.2 (macOS 10.15.4)
PlatformIO Core 4.3.3 Home 3.2.0 (Platform ststm32@~6.0.0, Board nucleo_f303re, Framework mbed)

#### How is this defect reproduced ? 

Initialize a I2C object and read some data from it.
Now change the AHBCLKDivider to an other value than RCC_SYSCLK_DIV1, for example RCC_SYSCLK_DIV8.
Now its impossible to communicate with the I2C device and receive valid values.

#### Possible solution?

Using the HAL_RCC_GetSysClockFreq() function for getting the current SysClock instead of using SystemCoreClock.

0xc0170, 2020-04-30:
cc @ARMmbed/team-st-mcd

jeromecoutant, 2020-05-06:
Hi

We agree that our current I2C porting doesn't allow other SystemCoreClock that the default one...

https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_STM/TARGET_STM32F3/i2c_device.h

Any help is welcomed

0xc0170, 2020-05-06:
@mrdernst would you be able to create a patch to fix this?

jeromecoutant, 2020-05-06:
Idea could be to study
https://github.com/STMicroelectronics/STM32CubeH7/blob/master/Drivers/BSP/STM32H747I-DISCO/stm32h747i_discovery_bus.c#L519

jeromecoutant, 2021-04-22:
Hi @mrdernst 
Could you have a look on #14557 ?
Thx

jeromecoutant, 2021-05-24:
@adbridge 
I think this issue could be closed with #14557 
Thx

closed 2021-05-24 by adbridge