ARMmbed/mbed-os #12912: KVStore: kvstore-general_tests_phase_2 failing with underestimated storage size

ccli8, 2020-05-04:
### Description of defect

On Nuvoton's targets, `features-storage-tests-kvstore-general_tests_phase_2` / `SEC_set_add_data_set_key_value_five_Kbytes` is failing. Looking into, the estimated storage size needed for `SecureStore` sees flashiap's program (page) size.

https://github.com/ARMmbed/mbed-os/blob/af4c8a94f3913ceda9069ec79cba81a793d695fe/features/storage/TESTS/kvstore/general_tests_phase_2/main.cpp#L87-L92

https://github.com/ARMmbed/mbed-os/blob/af4c8a94f3913ceda9069ec79cba81a793d695fe/features/storage/TESTS/kvstore/general_tests_phase_2/main.cpp#L127-L132

On Nuvoton targets, they are only 4-bytes, and cause underestimation.

https://github.com/ARMmbed/mbed-os/blob/af4c8a94f3913ceda9069ec79cba81a793d695fe/targets/TARGET_NUVOTON/TARGET_M261/flash_api.c#L90-L98

#### Target(s) affected by this defect ?

NUMAKER_IOT_M263A
NU_PFM_M2351_NPSA_NS

#### Toolchain(s) (name and version) displaying this defect ?

n/a

#### What version of Mbed-os are you using (tag or sha) ?

af4c8a94f3913ceda9069ec79cba81a793d695fe

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

n/a

#### How is this defect reproduced ? 

```sh
mbed test -m NUMAKER_IOT_M263A -t ARMC6 -n features-storage-tests-kvstore-general_tests_phase_2
```

0xc0170, 2020-05-04:
cc @ARMmbed/mbed-os-storage

kyle-cypress, 2020-05-04:
@ccli8 , what is the erase sector size on this device?
Are you able to tell how many pages would have been sufficient?

ccli8, 2020-05-05:
> what is the erase sector size on this device?
Are you able to tell how many pages would have been sufficient?

@kyle-cypress These Nuvoton targets have 2KiB erase sector size and only <b>4-byte</b> program page size. Not familiar with SecureStore, so I cannot give a estimation. But if I deliberately enlarge erase sector size from 2KiB to e.g. 8KiB below, the tests pass. The `program_size * PAGES_ESTIMATE` estimation may get unreasonable because `program_size` is only 4-byte on Nuvoton targets, I guess.

<pre>
ul_bd_size  = align_up(program_size * PAGES_ESTIMATE, <b>0x2000</b> * 2); 
</pre>

0xc0170, 2020-05-06:
I'll review this shortly as I noticed the nightly has very similar failures, will update.

0xc0170, 2020-05-06:
This should not be closed,I'l lreopen as the issue is still active (it was duplicated)

kyle-cypress, 2020-05-07:
> > what is the erase sector size on this device?
> > Are you able to tell how many pages would have been sufficient?
> 
> @kyle-cypress These Nuvoton targets have 2KiB erase sector size and only 4-byte program page size. Not familiar with SecureStore, so I cannot give a estimation. But if I deliberately enlarge erase sector size from 2KiB to e.g. 8KiB below, the tests pass. The `program_size * PAGES_ESTIMATE` estimation may get unreasonable because `program_size` is only 4-byte on Nuvoton targets, I guess.
> 
> ul_bd_size  = align_up(program_size * PAGES_ESTIMATE, 0x2000 * 2);

Yes, that would be my theory as well. Perhaps the size computation should use the lower of either program_size or the minimum size needed to hold a small key. I know that it can fit into a single 512 byte page, but I'm not sure whether 512 bytes would be a reasonable baseline or if the test should pick something smaller.

LDong-Arm, 2020-11-19:
Hi @ccli8, I fixed this issue in #13848 after I encountered the same test failure. The fix is https://github.com/ARMmbed/mbed-os/commit/13c5b64e16052a905b8eb6f6a243ad81afa1c9a3#diff-49a25f9feefaffecad0fcd30c50dc9331cff8b55ece53def6285c09e17e6f5d7. Would you please have a try?

ccli8, 2020-11-20:
@LDong-Arm #13848 can fix my failure. Thanks.

0xc0170, 2020-11-20:
I'll close as resolved.

closed 2020-11-20 by 0xc0170