ARMmbed/mbed-os #12921: mbed-cloud-client-example no longer works with kvstore FILESYSTEM

teetak01, 2020-05-05:
### Description of defect

https://github.com/armmbed/mbed-cloud-client-example

Fails to boot-up on K64F when using
`
            "storage.storage_type"                      : "FILESYSTEM",
`

kv_init_storage_config() - failed, status -2130771697

Reproducable with at least ethernet and Wi-Fi configurations.

Similar target K66F works, but it uses storage_type: TDB_INTERNAL so I presume the issue is related to the external storage.

#### Target(s) affected by this defect ?

K64F

#### Toolchain(s) (name and version) displaying this defect ?

GCC_ARM, ARMC6

#### What version of Mbed-os are you using (tag or sha) ?

#dbf05ceaed89

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

#### How is this defect reproduced ? 

CI automation.

0xc0170, 2020-05-05:
cc @ARMmbed/mbed-os-storage

teetak01, 2020-05-06:
With Mbed OS 6.0, the (client) application needs to call now explicitly DeviceKey::generate_root_of_trust().

This is due changes done in #12385 and #12823 

This will be implemented in future version of Device Management Client.

0xc0170, 2020-05-06:
@teetak01 Can we close this issue ? As its being tracked separately.

teetak01, 2020-05-06:
Yeah, for reference if someone finds this issue while testing with Mbed OS 6 and external KVstore:

This impacts the usage with external KVstore implementations (FILESYSTEM, TDB_EXTERNAL). The internal-flash only TDB_INTERNAL configuration is fine as that does not have DeviceKey dependencies.

closed 2020-05-06 by teetak01