ARMmbed/mbed-os #12922: mbed-cloud-client-example fatal run-time error on ST boards (SetSysClock failed)

teetak01, 2020-05-05:
### Description of defect

https://github.com/armmbed/mbed-cloud-client-example hits Fatal runtime failure on several ST boards.

++ MbedOS Error Info ++
Error Status: 0x80FF0100 Code: 256 Module: 255
Error Message: Fatal Run-time error
Location: 0x8053617
Error Value: 0x0
Current Thread: <unnamed> Id: 0x0 Entry: 0x80029C9 StackSize: 0x0 StackMem: 0x80029C9 SP: 0x2001FF68
For more info, visit: https://mbed.com/s/error?error=0x80FF0100&tgt=NUCLEO_F411RE
-- MbedOS Error Info --
SetSysClock failed

#### Target(s) affected by this defect ?

Nucleo F411RE, Nucleo F429ZI

#### Toolchain(s) (name and version) displaying this defect ?

GCC_ARM, ARMC6

#### What version of Mbed-os are you using (tag or sha) ?

#dbf05ceaed89

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

#### How is this defect reproduced ? 

Should be reproducable with the https://github.com/armmbed/mbed-cloud-client-example
using NUCLEO_F429ZI as target as described in tutorial:
https://www.pelion.com/docs/device-management/current/connecting/mbed-os.html

0xc0170, 2020-05-05:
cc @ARMmbed/team-st-mcd

jeromecoutant, 2020-05-05:
Hi
I could see that #12662 has been lost during #12626

adbridge, 2020-05-05:
@teetak01 looks like the sha has been rejected because it doesn't follow either the short form GitHub format or the full sha ?

teetak01, 2020-05-05:
Well, that is output from mbed ls

`|- mbed-os (#dbf05ceaed89)`

I think that should be acceptable format?

jeromecoutant, 2020-05-05:
Hi
In mbed-cloud-client-example, where does bootloader come from ?
  "target.bootloader_img" : "tools/mbed-bootloader-nucleo_f429zi-internal_flash-no_rot-v4.1.0.bin",

Thx

adbridge, 2020-05-05:
> 
> 
> Well, that is output from mbed ls
> 
> `|- mbed-os (#dbf05ceaed89)`
> 
> I think that should be acceptable format?

Mmmm interesting that would imply mbed ls is not providing a format normally used by github....

jeromecoutant, 2020-05-05:
It seems that after #12626 (and up to date drivers), system clock PLL can not be configured if PLL is already configured, which is the case with bootloader.

@LMESTM

0xc0170, 2020-05-05:
Thanks @jeromecoutant . Could this be fixed? We would like to have this in beta1 release (today all PR are in for beta1)

jeromecoutant, 2020-05-05:
Could you check #12924 ?

teetak01, 2020-05-06:
Hi @jeromecoutant I tested your fix and it works fine! Thanks for the quick response.

PR #12924 (2020-05-06): STM32F4 bypass PLL configuration when already done by bootloader
### Summary of changes 

Fix #12922 

It seems that after #12626 (and up to date drivers), system clock PLL can not be configured if PLL is already configured, which is the case with bootloader.

@ARMmbed/team-st-mcd 

#### Impact of changes 

#### Migration actions required 

### Documentation 

----------------------------------------------------------------------------------------------------------------
### Pull request type
closed 2020-05-06 by 0xc0170