ARMmbed/mbed-os #12930: Integer overflow in MbedOS CoAP library parser

mjurczak, 2020-05-05:
### Description of defect
**References:**

https://github.com/ARMmbed/mbed-os/tree/mbed-os-5.15.3/features/frameworks/mbed-coap

https://github.com/ARMmbed/mbed-coap/tree/v5.1.5

**File:**

[sn_coap_parser.c](https://github.com/ARMmbed/mbed-os/blob/b6370b4c37f3d4665ed1cdcb1afea85396bba1b3/features/frameworks/mbed-coap/source/sn_coap_parser.c)

**Analysis:**

Unhandled option length variable roll-over occurs if extended option length encoding is used with >65535 length encoded. The frame is further processed with the result of integer roll-over.

https://github.com/ARMmbed/mbed-os/blob/b6370b4c37f3d4665ed1cdcb1afea85396bba1b3/features/frameworks/mbed-coap/source/sn_coap_parser.c#L341-L354

**Type:**

- Integer Overflow or Wraparound

**Result:**

- Undetected malformed frame
- Incorrect packet parsing

#### Target(s) affected by this defect ?

- MbedOS mbed-coap library 5.1.5
- MbedOS 5.15.3

#### Toolchain(s) (name and version) displaying this defect ?
N/A

#### What version of Mbed-os are you using (tag or sha) ?
MbedOS 5.15.3

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
N/A

#### How is this defect reproduced ? 
N/A

0xc0170, 2020-05-06:
cc @teetak01

teetak01, 2020-05-06:
Thanks for the reports related to CoAP library. We will go through the findings.

mjurczak, 2020-05-06:
@teetak01 No problem. Let me know if I can assist with reproducing or suggesting fixes.

teetak01, 2020-05-07:
Thanks @mjurczak, mbed-coap is developed in https://github.com/ARMmbed/mbed-coap repository.

We would appreciate any contributions and/or feedback there.

mjurczak, 2020-05-11:
A solution proposal:
https://github.com/mjurczak/mbed-coap/commit/4647a68e364401e81dbd370728127d844f221d93

teetak01, 2020-05-19:
Hi @mjurczak we can review the fixes, but we would need you to open PRs vs [mbed-coap](https://github.com/ARMmbed/mbed-coap) first,

mjurczak, 2020-05-21:
Hi @teetak01, I understand. I will clean-up the code and commits a bit, and open new PRs targeting specific issues in mbed-coap repository.

closed 2020-07-02 by ciarmcom