ARMmbed/mbed-os #12932: IAR compilation issue with

jeromecoutant, 2020-05-06:
### Description of defect

Since #12761 3 tests cases can't be compiled any more:
- tests-mbed_hal-rtc
- tests-mbedmicro-rtos-mbed-event_flags
- tests-mbedmicro-rtos-mbed-signals

````
[Error] Callback.h@628,10: [Pe1574]: static assertion failed with "F must be TriviallyCopyable. Turn on Mbed configuration option 'platform.callback-nontrivial' to use more complex function objects"
````

#### Target(s) affected by this defect ?

All

#### Toolchain(s) (name and version) displaying this defect ?

IAR V8.32.3.193/W32 for ARM

#### What version of Mbed-os are you using (tag or sha) ?

master branch : 0df2188f3e9b8c90657f51b07d0aa5e87cafd4d3

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

NA

#### How is this defect reproduced ? 

NA

0xc0170, 2020-05-06:
cc @kjbracey-arm @ARMmbed/mbed-os-core

kjbracey-arm, 2020-05-06:
Apparently IAR may be within its rights here, at least for the event flags and signals.

They're using lambdas, and lambda closure types don't have to be trivial, even when it seems obvious that they should be:

> An implementation may define the closure type differently from what is described below provided this does not alter the observable behavior of the program other than by changing:
> * the size and/or alignment of the closure type,
> * **whether the closure type is trivially copyable (Clause 9),**
> * whether the closure type is a standard-layout class (Clause 9), or
> * whether the closure type is a POD class (Clause 9).

Stack Overflow discussion [here](https://stackoverflow.com/questions/32986193/when-is-a-lambda-trivial)

And there was me thinking ARMC5 was in error in reporting lambdas as not trivially copyable. Although no evidence was seen that they weren't, other than the type trait report.

We could just deactivate the optimisation for IAR, I guess.  I'd rather do that than stop using lambdas. Or we could do what we did for ARMC5, which is remove the assert altogether, assuming that "should-be-trivial" lambdas can be treated as such. That means we won't get any assert failures for any non-trivial types though. You'd have to rely on the problem being spotted by other toolchains.

However, I don't see what the RTC problem is. I can only see one use of Callback with a simple function there. Can you give any more detail?

(looping in @pan- for a new information nugget.)

jeromecoutant, 2020-05-25:
> We could just deactivate the optimisation for IAR

Done in #13015 

> However, I don't see what the RTC problem is. I can only see one use of Callback with a simple function there. Can you give any more detail?

You're right, it is not related:

````
        [Error] main.cpp@203,0: [Pe2486]: user-defined literal operator not found
        [DEBUG] Return: 2
        [DEBUG] Output:       for (auto init_val = RealTimeClock::time_point(100s); init_val < RealTimeClock::time_point(400s); init_val += 100s) {
        [DEBUG] Output:                                                      ^
````

kjbracey-arm, 2020-05-25:
main.cpp is presumably missing `using namespace std::chrono` or `using namespace std::chrono_literals`? Would have expected other toolchains to show problems though. Random toolchain-dependent header inclusions are expected, but not random namespace additions.

jeromecoutant, 2020-05-25:
> main.cpp is presumably missing `using namespace std::chrono` or `using namespace std::chrono_literals`?

no...
https://github.com/ARMmbed/mbed-os/blob/master/TESTS/mbed_hal/rtc/main.cpp#L34

kjbracey-arm, 2020-05-26:
`std::chrono` should transitively give you the literals via a `using` directive inside that namespace in `<chrono>`. And `<chrono>` must be included by the header that gave you `RealTimeClock`.

What if you add `using namespace std::chrono_literals` yourself? But this is far from the only file using `using namespace std::chrono` alone. Is all that other Chrono stuff working?

jeromecoutant, 2020-05-26:
> What if you add `using namespace std::chrono_literals` yourself?

I tried, but doesn't compile...

> Is all that other Chrono stuff working?

Yes...

kjbracey-arm, 2020-05-26:
> I tried, but doesn't compile...

???? Do you mean you replaced `chrono` with `chrono_literals`? I meant to add it.

Whole thing seems like a random compiler bug. There are plenty of other `s` literals in the file, above the one that's showing the error.

No idea what's going on. At the specific error point you could replace `100s` with `seconds(100)`?

jeromecoutant, 2020-05-26:
> ???? Do you mean you replaced `chrono` with `chrono_literals`? I meant to add it.

yes, replace or add gives same result

> No idea what's going on. At the specific error point you could replace `100s` with `seconds(100)`?

This is working!
````
$ git diff TESTS/mbed_hal/rtc/main.cpp
diff --git a/TESTS/mbed_hal/rtc/main.cpp b/TESTS/mbed_hal/rtc/main.cpp
index 947bae34e8..b3aa442018 100644
--- a/TESTS/mbed_hal/rtc/main.cpp
+++ b/TESTS/mbed_hal/rtc/main.cpp
@@ -32,6 +32,7 @@

 using namespace utest::v1;
 using namespace std::chrono;
+using namespace std::chrono_literals;

 static constexpr auto WAIT_TIME = 4s;
 static constexpr auto WAIT_TOLERANCE = 1s;
@@ -200,7 +201,7 @@ void rtc_write_read_test()
 {
     RealTimeClock::init();

-    for (auto init_val = RealTimeClock::time_point(100s); init_val < RealTimeClock::time_point(400s); init_val += 100s) {
+    for (auto init_val = RealTimeClock::time_point(seconds(100)); init_val < RealTimeClock::time_point(400s); init_val += 100s) {
         core_util_critical_section_enter();

         RealTimeClock::write(init_val);
````

jeromecoutant, 2020-05-26:
RealTimeClock::time_point needs a const ?

kjbracey-arm, 2020-05-26:
> This is working!

Well, that's both changes. Which one(s) fixed it?

> RealTimeClock::time_point needs a const ?

Not sure what you mean. I guess a constructor taking `duration &` instead of `const duration &`?

`RealTimeClock::time_point` is just a "typedef": `using time_point = std::chrono::time_point<RealTimeClock>;`.

So we're invoking the duration-taking constructor of a `std::chrono::time_point` template instantiation. That should be `const duration &`.

This might be the only place in the codebase where we actually do use that constructor with a literal, Not sure. Might explain why we're not seeing the flaw somewhere else. Other places we're probably passing variables, or non-constant expressions.

Another way of phrasing it, sidestepping that constructor overload, would be `RealTimeClock::time_point() + 100s`. That's pretty neat, try it.

kjbracey-arm, 2020-05-26:
The daft thing is you've still got `RealTimeClock::time_point(400s)` on the same line...

kjbracey-arm, 2020-05-26:
Could be that the `auto` context is throwing it, but not sure why that should matter. Yet another way of writing it is:

    for (RealTimeClock::time_point init_val{100s}; init_val < RealTimeClock::time_point(400s); init_val += 100s) {

Bit odd to not use a `=` initialisation in a `for`, but legal. Saves the repetition that `auto` was trying to avoid:

    for (RealTimeClock::time_point init_val = RealTimeClock::time_point(100s); init_val < RealTimeClock::time_point(400s); init_val += 100s) {

kjbracey-arm, 2020-05-26:
I wonder if `{}` versus `()` on the constructor invocation matters...

jeromecoutant, 2020-05-26:
Other propositions don't work, but as the "seconds(100)" is OK, I stop... :-)

PR #13015 (2020-06-05): Force callback non-trivial for IAR
### Summary of changes 

This fixes #12932 

#12761 optimisation is disabled for IAR

#### Impact of changes 

IAR tests are now OK

#### Migration actions required 

### Documentation 

----------------------------------------------------------------------------------------------------------------
### Pull request type 

    [x] Patch update (Bug fix / Target update / Docs update / Test update / Refactor)
    [] Feature update (New feature / Functionality change / New API)
    [] Major update (Breaking change E.g. Return code change /
closed 2020-06-05 by 0xc0170