ARMmbed/mbed-os #12936: MultiTech Dragonfly F413RH modem fails to reconnect after reset

maclobdell, 2020-05-07:
### Description of defect

While running Mbed OS network integration testing and Pelion Device Management testing I observed that the network would fail to reconnect after re-flashing the application. It was resolved by power cycling the board.

It seems that any reset of the MCU will cause the communication to the cellular network modem to hang. 
See attached logs.
[dragonfly_f413_trace_log.txt](https://github.com/ARMmbed/mbed-os/files/4590475/dragonfly_f413_trace_log.txt) See line 977.
[dragonfly-f413rh-integration-tests-gcc-master-30march2020.txt](https://github.com/ARMmbed/mbed-os/files/4590477/dragonfly-f413rh-integration-tests-gcc-master-30march2020.txt)

#### Target(s) affected by this defect ?
MultiTech Dragonfly F413RH

#### Toolchain(s) (name and version) displaying this defect ?
All

#### What version of Mbed-os are you using (tag or sha) ?
Mbed OS 5.15.3

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
n/a

#### How is this defect reproduced ? 

Executing all integration tests results in some network tests that fail
mbed test -m MTS_DRAGONFLY_F413RH -t GCC_ARM -n tests-integration-* -DINTEGRATION_TESTS

However, if they run individually, they pass.
   mbed test -m MTS_DRAGONFLY_F413RH -t GCC_ARM -n tests-integration-net-single* -DINTEGRATION_TESTS

#### Fix for 5.15.x
Leon and MultiTech shared a fix that works with 5.15.x. 
replace this file at mbed-os\targets\TARGET_STM\TARGET_STM32F4\TARGET_STM32F413xH\TARGET_MTS_DRAGONFLY_F413RH
[onboard_modem_api.c.txt](https://github.com/ARMmbed/mbed-os/files/4590485/onboard_modem_api.c.txt)

**An equivalent fix is needed for Mbed OS 6, which has some relevant functions refactored**

@ARMmbed/team-multitech - please submit a PR to fix.

felser, 2020-05-08:
I made the change to mbed6 and confirmed that it was failing before and passing after my changes were added. However, the integration tests still mostly fail but it doesn’t look like that is due to anything specific to the Dragonfly. I enabled trace logging with AT debug then ran: 
mbed test -t GCC_ARM -m MTS_DRAGONFLY_F413RH -n mbed-os-tests-integration-n* -DINTEGRATION_TESTS –v
(notice integration-n*)
Only two sets of tests run but the second set passes only if my code is in place. I see a message “Modem power ON” in both sets of tests. With my code in place, the radio responds after "Modem power ON" while without it, there is no response. I know reviewers like to see test results. Can I get by with this basic description or do I need a log? The log has a lot of information printed but I could strip out superfluous messages.

To summarize:
1.	What do I NEED to provide as proof that the fix is good for this PR?
2.	Why are the integration tests failing so badly? Is this just a work in process with mbed6?

maclobdell, 2020-05-09:
If possible, please share a test log.

IMPORTANT NOTICE: The contents of this email and any attachments are confidential and may also be privileged. If you are not the intended recipient, please notify the sender immediately and do not disclose the contents to any other person, use it for any purpose, or store or copy the information in any medium. Thank you.

maclobdell, 2020-07-17:
fixed by https://github.com/ARMmbed/mbed-os/pull/12966

closed 2020-07-17 by maclobdell