ARMmbed/mbed-os #12952: WB55RG returns wrong erase sector size

micgur01, 2020-05-11:
### Description of defect

FlashIAPBlockDevice
get_sector_size return 4096B, instead of 128B

#### Target(s) affected by this defect ?

NUCLEO_WB55RG

#### Toolchain(s) (name and version) displaying this defect ?

TOOLCHAIN_GCC_ARM

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-5.12

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

mbed-cli --version
1.10.2

arm-none-eabi-gcc --version
arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 8-2019-q3-update) 8.3.1 20190703 (release) [gcc-8-branch revision 273027]

#### How is this defect reproduced ? 

calling an API FlashIAPBlockDevice

jeromecoutant, 2020-05-11:
Hi
Why are you using mbed-os-5.12 ?
Could you check master branch ? :-)

LMESTM, 2020-05-12:
> ### Description of defect
> FlashIAPBlockDevice
> get_sector_size return 4096B, instead of 128B

Which sector are your referring to ? 
Each page size of STM32WB55RG is 4KB , I don't think there is any area of 128 bytes, which sector address are you referring to ?

LMESTM, 2020-05-13:
@micgur01  not sure if you've seen above comment ? I think that 4KB is the atcual sector size of STM32WB so unless we get more details I would tend to reject this issue  ...

micgur01, 2020-05-13:
I understand , also checked this issue.
It has indeed 4KB size.

LMESTM, 2020-05-13:
@micgur01 - so will you close the issue then ?

closed 2020-05-16 by micgur01