ARMmbed/mbed-os #12953: Application stuck after erasing 4k sector NUCLEO_WB55RG at boot

micgur01, 2020-05-11:
Description of defect

Need to erase sector at boot, 
But get_sector_size return 4096B, instead of 128B erasing it at boot , the application get stuck

Target(s) affected by this defect ?
NUCLEO_WB55RG

Toolchain(s) (name and version) displaying this defect ?
TOOLCHAIN_GCC_ARM

What version of Mbed-os are you using (tag or sha) ?
mbed-os-5.12

What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
mbed-cli --version
1.10.2

arm-none-eabi-gcc --version
arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 8-2019-q3-update) 8.3.1 20190703 (release) [gcc-8-branch revision 273027]

How is this defect reproduced ?
Erasing sector at application start.

0xc0170, 2020-05-12:
How this relates to #12952 - if its fixed, will this be as well or what issue is reported here?

jeromecoutant, 2020-06-08:
Could we close this one as well (like #12952) ?
Thx

micgur01, 2020-06-08:
at application starts, erasing one sector 4k.
it takes time...
I suppose at the same time BLE starts and erasing sector interrupts the BLE startup.
BLE get stuck - no connection afterwards.

jeromecoutant, 2020-07-02:
> What version of Mbed-os are you using (tag or sha) ?
> mbed-os-5.12

You definitively have to try master branch...
Thx

jeromecoutant, 2020-07-17:
#13304 is implementing the last flash procedure according to ST BLE experts.

0xc0170, 2020-08-21:
> #13304 is implementing the last flash procedure according to ST BLE experts.

I'll close this as resolved.

closed 2020-08-21 by 0xc0170