ARMmbed/mbed-os #12974: Shared event queue not working as expected

marcuschangarm, 2020-05-14:
### Description of defect

Based on this documentation: https://os.mbed.com/docs/mbed-os/v5.15/apis/eventqueue.html#shared-event-example-deferring-from-interrupt-context , I was expecting to be able to post events to the shared event queue using the code below.

#### Target(s) affected by this defect ?

Tested on:
* K64F
* CY8CPROTO_062_4343W

#### Toolchain(s) (name and version) displaying this defect ?

GCC 9.2.1

#### What version of Mbed-os are you using (tag or sha) ?

Tested on:
* master (06ff4cfb81dc20474a83bd6033b3200175af5e58)
* mbed-os-5.15.3
* mbed-os-5.14.2

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

mbed-cli 1.10.2

#### How is this defect reproduced ? 

Example code:

```
#include "mbed.h"
#include "mbed_events.h"

void hello(void)
{
    printf("hello\r\n");
}

int main() {
    EventQueue *queue = mbed_event_queue();

    printf("Queue %p\r\n", queue);

    for (;;) {
        printf("schedule\r\n");

        queue->event(hello);
        ThisThread::sleep_for(1000);
    }
}
```

0xc0170, 2020-05-15:
cc @ARMmbed/mbed-os-core @kjbracey-arm

kjbracey-arm, 2020-05-15:
`Queue::event` makes an `Event` function-like object which can be used to post the event when invoked later.

You would use it like this:

    // This allocates space for the event in the event queue
    auto e = queue->event(hello);

    e();  // This queues the event.

In that example, it's passing the `Event` to `InterruptIn::fall`, so that when the interrupt happens, the event gets posted.

If you want to just make and queue the event immediately, rather preparing it to call later, you do:

    queue->call(hello);

Check out the EventQueue tutorial page linked at the bottom of that page - it breaks it down more than those examples. I think that tutorial should be a more prominent link.

evedon, 2020-05-15:
I think the last example https://os.mbed.com/docs/mbed-os/v5.15/apis/eventqueue.html#static-eventqueue-example-posting-user-allocated-events-to-the-static-queue is particularly useful. We should perhaps re-organise the page to give it more emphasis.

kjbracey-arm, 2020-05-15:
Main issue I think with EventQueue is that you kind of need to understand the basic model first - the examples themselves are fine, but trying to figure out what's going on just from the examples you see on that first page is hard. Those are examples of pretty much everything you can do, which is a lot... People should start from the tutorial.

marcuschangarm, 2020-05-15:
Ah, I see the difference, thank you for the explanation!

closed 2020-05-15 by marcuschangarm