ARMmbed/mbed-os #12975: Mbed isn't recognizing the total SRAM on TARGET_NUCLEO_WB55RG

AlefeFelipe, 2020-05-14:
### Description of defect

The mbed isn't recognizing the total SRAM on NUCLEO WB55RG. It has two SRAM, one of size 192kb, and other of size 64KB with parity check. But the result from compilation and system info debug results only in 192KB as total.
Result from compilation:
```
Total Static RAM memory (data + bss): 119272(+0) bytes
Total Flash memory (text + data): 615548(+0) bytes

```
Result from system info debug:
```
================= CPU STATS =================
Idle: 91% Usage: 9% Peak: 67%
================ HEAP STATS =================
Heap size: 54971 / 75936 bytes
================ THREAD STATS ===============
Current Thread: memory_monitor                 Thread: 0x2001A818 	 Stack size: 872 / 4096
Current Thread: rtx_idle                       Thread: 0x20016D08 	 Stack size: 280 / 768
Current Thread: shared_event_queue             Thread: 0x200023E4 	 Stack size: 192 / 2048
Current Thread: coap_thread                    Thread: 0x20019AFC 	 Stack size: 176 / 2048
Current Thread: thread1                        Thread: 0x2001AF70 	 Stack size: 184 / 1280
Current Thread: thread2                        Thread: 0x20018EE8 	 Stack size: 1336 / 8192
Current Thread: thread3                        Thread: 0x2001944C 	 Stack size: 848 / 4096
Current Thread: thread4                        Thread: 0x2001AC14 	 Stack size: 744 / 2048
Current Thread: rtx_timer                      Thread: 0x20016CC4 	 Stack size: 104 / 768
Current Thread: stm_queue                      Thread: 0x20020F78 	 Stack size: 984 / 2048
Current Thread: thread5                        Thread: 0x2001A484 	 Stack size: 1072 / 2048
Current Thread: main                           Thread: 0x20016BB8 	 Stack size: 1792 / 8192
Current Thread: ppp                            Thread: 0x20021B78 	 Stack size: 684 / 816
Current Thread: thread6                        Thread: 0x20019A3C 	 Stack size: 8224 / 16384
Current Thread: application_unnamed_thread     Thread: 0x2001D690 	 Stack size: 464 / 1024
Current Thread: nanostack_event_thread         Thread: 0x200156B8 	 Stack size: 1400 / 8192
Current Thread: shared_highprio_event_queue    Thread: 0x20001DBC 	 Stack size: 368 / 1024
```
The sum is: 75936 + 119272 = 195208

The reason for this can probably be the information on the file index.json, inside mbed-os/tools/arm_pack_manager, where is pointed that STM32WB55RGVx has only one memory bank, despite the fact that it was configured as 256kb SRAM, maybe that's the reason.

If I'm doing something wrong can someone please help?

#### Target(s) affected by this defect ?
NUCLEO_WB55RG

#### Toolchain(s) (name and version) displaying this defect ?
GCC_ARM

#### What version of Mbed-os are you using (tag or sha) ?

5.15.2
#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
mbed-cli - version 1.10.1
#### How is this defect reproduced ? 
Enable mbed memory stats on the mbed_app.json and program the NUCLEO_WB55RG board, sum the HEAP size and compilation size to get total
[image]

jeromecoutant, 2020-05-15:
Hi

SRAM1 is dedicated for M4 core, and then for mbed-os applications.
SRAM2 is dedicated for M0 core and inter CPU communication, and then can not be addressed.

I need to update
https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_STM/TARGET_STM32WB/README.md

Thx

0xc0170, 2020-05-15:
Once updated, let us know, we shall close this?

AlefeFelipe, 2020-05-15:
Ok, thanks for answering

closed 2020-05-15 by AlefeFelipe