ARMmbed/mbed-os #12976: Remove Mutex::_count

AriParkkila, 2020-05-15:
### Description of defect

[Mutex::_count](https://github.com/ARMmbed/mbed-os/blob/b1071487a3a68fcb39397350bf15bdebaa8a4ec4/rtos/Mutex.h#L180) should be removed or implemented properly.

`Mutex` class is not actually using `_count` for anything, but because it's [updated without holding a lock](https://github.com/ARMmbed/mbed-os/blob/b1071487a3a68fcb39397350bf15bdebaa8a4ec4/rtos/source/Mutex.cpp#L128) that breaks implementation of the friend class [ConditionVariable](https://github.com/ARMmbed/mbed-os/blob/b1071487a3a68fcb39397350bf15bdebaa8a4ec4/rtos/source/ConditionVariable.cpp#L61).

I guess the original idea behind `_count` was to implement recursive mutex, but it's not really needed as that's already handled (properly) in `rtx_mutex.c`. Also if `ConditionVariable` need to guarantee `Mutex` is non-recursive then simply deassert for `osMutexRecursive` attribute.

#### Target(s) affected by this defect ?

All with `rtos.present`.

#### Toolchain(s) (name and version) displaying this defect ?

All.

#### What version of Mbed-os are you using (tag or sha) ?

https://github.com/ARMmbed/mbed-os/commit/0c4e50abb3fa1ec3d80f19c88ceefa106ff94e64

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

#### How is this defect reproduced ? 

E.g. this code:
```
    Mutex m;
    Thread t;
    t.start([&m]() {
        while (1) {
            m.lock();
            m.unlock();
        }
    });

    ConditionVariable cv(m);
    while (1) {
        m.lock();
        cv.wait_for(0);
        m.unlock();
    }
```
... eventually it crashes:
```
++ MbedOS Error Info ++
Error Status: 0x80FF0144 Code: 324 Module: 255
Error Message: Assertion failed: _mutex._count == 1
Location: 0x1DEAF
File: ./mbed-os/rtos/source/ConditionVariable.cpp+52
Error Value: 0x0
Current Thread: main Id: 0x20004700 Entry: 0x1F3DB StackSize: 0x1FA0 StackMem: 0x200053F8 SP: 0x200071D4 
For more info, visit: https://mbed.com/s/error?error=0x80FF0144&tgt=MTB_LAIRD_BL654
-- MbedOS Error Info --
```

0xc0170, 2020-05-15:
@kjbracey-arm @ashok-rao Was  this discussed  this week in #12965 and #12964 ?

ashok-rao, 2020-05-15:
@0xc0170 .. Nop. Looks like @AriParkkila came across "_count" and Mutex the same time we did, but our requirements are different..

kjbracey-arm, 2020-05-15:
Nice test. Weird lower loop, but maybe it's necessary to provoke failure. Hope you're not using it like that in practice. It should look more like

    ConditionVariable cv(m);
    m.lock();
    while (1) {
        cv.wait_for(0);
    }

to make sure the `while` condition is tested with the mutex held. (The mutex isn't just some random thing you have to do around the wait call for no reason - it's to make the test-and-wait combo atomic).

Looks like I broke it in b8e80dd2fb3fac8d8e0e2 (#10358) 

Previous behaviour was always to decrement count inside the lock, before the unlock call. That was safe, but would have left it wrong in the event of any error, which is probably what I meant to fix, but that's not a big deal. Errors mean system death anyway.

The original reason for it was purely to implement the `ConditionVariable` assert.

I argued against the `_count` when it originally went in (#3648), much like I've just been arguing against `Semaphore::get_count`, and got it demoted to just friend access to ConditionVariable. But taking it out now would be a potentially breaking change. I think we just fix the #10358 breakage, putting the decerement back where it was.

AriParkkila, 2020-05-18:
Thanks, fixed in #12983.

PR #12983 (2020-06-04): Mutex::unlock - decrement _count inside lock

### Summary of changes 

`Mutex::unlock` was decrementing the `_count` member after releasing the mutex, which meant it was unprotected, exposing a race that could corrupt the count.

This could lead to an assert in `ConditionVariable::wait`, which checks that the mutex count is one.

Fixes #12976. Bug was introduced in #10358 (5.13 and 5.12.2).

#### Impact of changes
closed 2020-05-18 by AriParkkila