ARMmbed/mbed-os #12980: release_version removal in targets causing tools to not report them as official targets

0xc0170, 2020-05-15:
### Description of defect

The function https://github.com/ARMmbed/mbed-os/blob/master/tools/build_api.py#L403, `get_mbed_official_release` is used by tools and who knows at the moment where else to be checked if a target is supported. As there are tools out there for 2 or 5, they most likely use this function. In our case, CI has not been updated, that should be done. But would not fix the issue in the other tools as well. If they do `get_mbed_official_release('5')`, only subset of targets is being return. Means we have broken it silently.

As a fix now, we shall fix the function  `get_mbed_official_release` to return all targets that are currently in targets.json file (so would be valid for Mbed 6). There are two use cases, ask for `2` or `5`. For 2, empty target list - not supported anymore. For `5`, return all targets as all are supported. This should fix the CI issue we are seeing and also any other tool that rely on this.

#### Target(s) affected by this defect ?

All targets that have no `release_version` attribute in targets.json file

#### Toolchain(s) (name and version) displaying this defect ?

all

#### What version of Mbed-os are you using (tag or sha) ?

77d55836495fabe649039255ac5f97b93f6d6b94

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

Not relevant

#### How is this defect reproduced ? 

Not easy as mbed cli does not use it as I was told.

cc @Patater @ARMmbed/mbed-os-core

0xc0170, 2020-05-15:
My quick  fix proposal:

```
diff --git a/tools/build_api.py b/tools/build_api.py
index 8da1666d48..b7c3a2158b 100755
--- a/tools/build_api.py
+++ b/tools/build_api.py
@@ -411,26 +411,25 @@ def get_mbed_official_release(version):
               RELEASE_VERSIONS
     """

+    # we ignore version for Mbed 6 as all targets in targets.json file are being supported
+    # if someone passes 2, we return emtpy tuple, if 5, we keep the behavior the same as
+    # release version is deprecated and all targets are being supported that are present
+    # in targets.json file
+
+    if version == `2`:
+        raise InvalidReleaseTargetException("Mbed 2 not supported anymore with this version of tools")
+
     mbed_official_release = (
         tuple(
             tuple(
                 [
                     TARGET_MAP[target].name,
-                    tuple(transform_release_toolchains(
-                        TARGET_MAP[target], version))
+                    tuple('ARM', 'GCC_ARM')
                 ]
-            ) for target in TARGET_NAMES \
-            if (hasattr(TARGET_MAP[target], 'release_versions')
-                and version in TARGET_MAP[target].release_versions)
+            ) for target in TARGET_NAMES
         )
     )

-    for target in mbed_official_release:
-        is_official, reason = is_official_target(target[0], version)
-
-        if not is_official:
-            raise InvalidReleaseTargetException(reason)
-
     return mbed_official_release

 def target_supports_toolchain(target, toolchain_name):
```

0xc0170, 2020-05-15:
https://github.com/ARMmbed/mbed-os/pull/12982 fix proposed

MarceloSalazar, 2020-05-15:
Note this PR https://github.com/ARMmbed/mbed-os/pull/12984 is related to remove "2" from `release_version` as there is no Mbed 2 support in the Mbed OS 6 code base.

0xc0170, 2020-05-20:
It was fixed via 12982 and also on CI side. I'll close this as resolved

closed 2020-05-20 by 0xc0170