ARMmbed/mbed-os #12989: Greentea FPGA testing fails with DISCO_H747I

umbynos, 2020-05-18:
### Description of defect
FPGA ci test shield fails most of the tests with DISCO_H747I

I'm trying to understand how FPGA testing works. I've cloned mbed-os repo, installed mbed-cli and run the generic tests with `mbed-cli test -t GCC_ARM --greentea -m auto` and everything is OK.
I then added **FPGA_CI_TEST_SHIELD** in `targets.json` as stated [there](https://os.mbed.com/docs/mbed-os/v5.15/porting/porting-targets.html). But after connecting the FPGA CI shield and running `mbed-cli test -t GCC_ARM --greentea -n tests-mbed_hal_fpga_ci_test_shield-spi -v` some tests are failing.
I've attached the execution output
[test_execution_fpga_spi.log](https://github.com/ARMmbed/mbed-os/files/4643673/test_execution_fpga_spi.log)
The only test working is pwm one. Every other FPGA test shows similar problems.

#### Target(s) affected by this defect ? 
As far as I know DISCO_H747I

#### Toolchain(s) (name and version) displaying this defect ?
GCC_ARM 9.2.1

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os (#475621ce36bf)

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
mbed-cli 1.10.2

#### How is this defect reproduced ? 
`mbed-cli test -t GCC_ARM --greentea -n tests-mbed_hal_fpga_ci_test_shield-spi -v` with FPGA CI shield attached

0xc0170, 2020-05-19:
@ARMmbed/team-st-mcd Please review

0xc0170, 2020-06-02:
I updated the SHA.

@ARMmbed/team-st-mcd Please review. Are you able to reproduce these reported failures?

jeromecoutant, 2020-06-03:
> @ARMmbed/team-st-mcd Please review. Are you able to reproduce these reported failures?

yes.... :-(

jeromecoutant, 2020-06-05:
Hi
Could you try #13073 ?
Regards,

umbynos, 2020-06-11:
Actually the problem it's not solved for me.

jeromecoutant, 2020-06-11:
Yes, it's good to have merge #13073, but @adbridge could you keep this issue opened
Thx

adbridge, 2020-06-11:
@jeromecoutant it seems that was a Github automatic response because I merged #13073 and they were linked ! I will reopen it.

PR #13073 (2020-06-11): STM32H7: FPGA tests support
### Summary of changes 

FPGA and CI-test-shield tests have been executed with NUCLEO_H743ZI2 and DISCO_H747I

- analogic dual pad support
- few pins definition updates

Fix #12989 

@ARMmbed/team-st-mcd 

#### Impact of changes 

#### Migration actions required 

### Documentation 

----------------------------------------------------------------------------------------------------------------
### Pull request type