ARMmbed/mbed-os #12990: STM32 target inherit: US_TICKER_MASK not defined

pilotak, 2020-05-18:
### Description of defect
When compiling a file below with target `-m NUCLEO_F412ZG` it compiles fine, but if I compile with `-m test` (custom target) the error below comes up.

```
Compile [ 99.0%]: hal_tick_overrides.c
[Error] hal_tick_overrides.c@53,56: 'US_TICKER_MASK' undeclared (first use in this function)
[ERROR] .\mbed-os\targets\TARGET_STM\hal_tick_overrides.c: In function 'HAL_GetTick':
.\mbed-os\targets\TARGET_STM\hal_tick_overrides.c:53:56: error: 'US_TICKER_MASK' undeclared (first use in this function)
   53 |     uint32_t elapsed_time = (((new_time - prev_time) & US_TICKER_MASK) + prev_tick_remainder);
      |                                                        ^~~~~~~~~~~~~~
.\mbed-os\targets\TARGET_STM\hal_tick_overrides.c:53:56: note: each undeclared identifier is reported only once for each function it appears in
```

#### Target(s) affected by this defect ?
Possibly all STM targets

#### Toolchain(s) (name and version) displaying this defect ?
GCC-ARM: 9.2.1 20191025

#### What version of Mbed-os are you using (tag or sha) ?

e345542b7005b9cc548e3c869f08813791ada442

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
mbed-cli: 1.10.1

#### How is this defect reproduced ? 
`main.cpp`
```cpp
#include "mbed.h"

int main() {
    printf("test\n");
    return 0;
}
```
`custom_targets.json`
```json
{
    "test": {
        "inherits": ["NUCLEO_F412ZG"],
        "supported_form_factors": [],
        "overrides": {
            "clock_source": "USE_PLL_HSE_XTAL"
        }
    }
}
```

pilotak, 2020-05-18:
It's only ok up to no later than d058586c482e084e920c37cbeae54d8782c88353

pilotak, 2020-05-28:
ping @jeromecoutant

jeromecoutant, 2020-05-28:
It's working on my side...

jeromecoutant, 2020-06-08:
@pilotak Do you find the issue on your side ?

pilotak, 2020-06-09:
Sorry for delay. I did an investigation and it only works if there in no `device.h` in `TARGET_TEST` folder.

`device.h` 
```cpp
#ifndef MBED_DEVICE_H
    #define MBED_DEVICE_H
    #include "objects.h"

    #define CONF_BOARD_TYPE     1
    #define CONF_BOARD_REV      3
    #define VOLTAGE_REFERENCE   12100000UL
    #define PROCESSOR_TYPE      1
#endif
```

jeromecoutant, 2020-06-09:
Good catch

Could you check https://github.com/jeromecoutant/mbed/commit/47407759a480ed9d3bafe5ea440d95b2e4bff8e5

Thx

pilotak, 2020-06-09:
yep, that fixed it

closed 2020-06-10 by 0xc0170

kjbracey-arm, 2021-04-13:
Fix in #13090 is invalid, and I'm requesting it be reverted.

The correct fix here would have been to ensure that you have the necessary include of us_ticker_defines.h in your replacement device.h, as the original device.h did.

Or make your device.h include the original device.h, rather than objects.h, then add your defines.

(I'm not actually clear on include path set-up here - if you have your own device.h, is it deterministic whether TARGET_TEST/device.h or TARGET_STM/device.h gets read?)

PR #13090 (2020-06-10): STM: move us_ticker_defines.h include
### Summary of changes 

Fix #12990 

"us_ticker_defines.h" include was strangely set in device.h file

@pilotak 

#### Impact of changes 

none

#### Migration actions required 

### Documentation 

----------------------------------------------------------------------------------------------------------------
### Pull request type 

    [x] Patch update (Bug fix / Target update / Docs update / Test update / Refactor)
    [] Feature update (New feature / Functionality change / New API)
    [] Major update (Breaking change E.g. Re