ARMmbed/mbed-os #12993: STM32F412 SetSysClock failed

pilotak, 2020-05-18:
### Description of defect

Exacly the same problem as in #12922 but because #12924 only fixes Nucleo F411RE and Nucleo F429ZI, it fails on other target(s) 

I applied the same fix as in PR locally and that solved it.

#### Target(s) affected by this defect ?
NUCLEO_F412ZG but all other should be checked in F4 family

#### Toolchain(s) (name and version) displaying this defect ?
GCC-ARM: 9.2.1 20191025

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-6.0.0-beta-1

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
mbed-cli: 1.10.1

#### How is this defect reproduced ? 
use bootloader and main app and it will fail.

jeromecoutant, 2020-05-19:
Hi

> I applied the same fix as in PR locally and that solved it.

Please, propose the pull request

Thx

jeromecoutant, 2020-06-05:
Could be closed thanks to #12996

closed 2020-06-05 by pilotak