ARMmbed/mbed-os #12994: pip install failed when install build enviroment

kabirz, 2020-05-19:
mbed-os  ➦ bf6f2c3c64  **pip install -r requirements.txt**
Defaulting to user installation because normal site-packages is not writeable
Collecting git+https://github.com/armmbed/manifest-tool.git@v1.4.6 (from -r requirements.txt (line 19))
  Cloning https://github.com/armmbed/manifest-tool.git (to revision v1.4.6) to /tmp/pip-req-build-j91rsdpn
  Running command git clone -q https://github.com/armmbed/manifest-tool.git /tmp/pip-req-build-j91rsdpn
  Running command git checkout -q 711dea922b36150e5f32c7d168cc20036b72a1af
    ERROR: Command errored out with exit status 1:
     command: /usr/bin/python3 -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-req-build-j91rsdpn/setup.py'"'"'; __file__='"'"'/tmp/pip-req-build-j91rsdpn/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /tmp/pip-pip-egg-info-m6uh9rf8
         cwd: /tmp/pip-req-build-j91rsdpn/
    Complete output (12 lines):
    Traceback (most recent call last):
      File "/tmp/pip-req-build-j91rsdpn/setup.py", line 28, in <module>
        from pip._internal.download import PipSession
    ModuleNotFoundError: No module named 'pip._internal.download'
    
    During handling of the above exception, another exception occurred:
    
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-req-build-j91rsdpn/setup.py", line 30, in <module>
        from pip.req import parse_requirements
    ModuleNotFoundError: No module named 'pip.req'
    ----------------------------------------
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.

kabirz, 2020-05-19:
work environment: Ubuntu 20.04 LTS

kabirz, 2020-05-19:
pip 19.3 is ok.

0xc0170, 2020-05-19:
What  was your previous pip version?

kabirz, 2020-05-20:
default version is the latest 20.1

kabirz, 2020-05-20:
The issue appear DAPLink(https://github.com/ARMmbed/DAPLink/blob/master/mbed-os.lib), latest mbed os is not, maybe you need update DAPLink.

adbridge, 2020-06-12:
@kabirz when raising issues in the future, please do not delete the issue template and instead actually fill in all the required fields. Thanks.