ARMmbed/mbed-os #12999: Statically allocated SPIFBlockDevice causes crash

rblaze, 2020-05-20:
### Description of defect
Here is a simple repro program:
```
#include <SPIFBlockDevice.h>
#include <mbed.h>

SPIFBlockDevice spif{PB_5, PB_4, PB_3, PA_15};

int main() {
  printf("Never reaching this point\n");
  return 0;
}
```
When I flash it, it crashes:
```
--- Terminal on /dev/ttyACM0 - 9600,8,N,1 ---
--- [RESET] ---

++ MbedOS Error Info ++
Error Status: 0x80010133 Code: 307 Module: 1
Error Message: Mutex: 0x20000D5C, Unknown
Location: 0x8002D79
Error Value: 0x20000D5C
Current Thread: main Id: 0x200010D0 Entry: 0x8002C9D StackSize: 0x1000 StackMem: 0x20001868 SP: 0x20004F3C 
For more info, visit: https://mbed.com/s/error?error=0x80010133&tgt=NUCLEO_F103RB
-- MbedOS Error Info --
```

Everything works when I comment out SPIFBlockDevice.
I'm using this board: https://os.mbed.com/users/hudakz/code/MapleMini_Hello/
There is the flash chip connected to the MCU, and it works when I'm using mbedos 5 or 6-alpha-3.

#### Target(s) affected by this defect ?
NUCLEO_F103RB

#### Toolchain(s) (name and version) displaying this defect ?
GCC_ARM, gcc-arm-none-eabi-9-2019-q4-major

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-6.0.0-beta-1

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
mbed-cli 1.10.2

#### How is this defect reproduced ? 
Build and flash the program above. It crashes 100% of time.

kjbracey-arm, 2020-05-22:
Issue #12981?

rblaze, 2020-05-23:
This helps, thanks.

closed 2020-05-23 by rblaze