ARMmbed/mbed-os #13000: u-blxo SARA-R142M unable to connect using LTE Cat M1

jdbruijn, 2020-05-20:
### Description of defect

I'm using a u-blox SARA-R412M module and was unable to get a connection to a LTE Cat M1 network. We have our own APN, which I've configured using `CellularContext::set_credentials` before starting the connection.

There appear to be two issues, which might be related so I'm listing both here. If they're not related, please let me know and I can split this into two separate issues.

1. I had to manually do the AT command `CGDCONT` in to be able to send data.
2. I had to add `CellularNetwork::C_REG` to the `reg_params._type` check in `CellularStateMachine::cellular_event_changed` in order to get a registration to the network.

I've attached a couple of logs that are, hopefully explaining, this behaviour. In the logs and diff I've redacted the custom APN with XXX as well as the IP's.

| log | `CGDCONT` set | `C_REG` added |
| --- | ------------- | ------------- |
| 569 | yes           | yes           |
| 570 | no            | yes           |
| 571 | yes           | no            |

Logs 569 and 571 are using the following patch in order to manually add the AT command `CGDCONT`, to fix 1.

```diff
diff --git a/features/cellular/framework/AT/AT_CellularNetwork.cpp b/features/cellular/framework/AT/AT_CellularNetwork.cpp
index 4e35dcb4..24910a61 100644
--- a/features/cellular/framework/AT/AT_CellularNetwork.cpp
+++ b/features/cellular/framework/AT/AT_CellularNetwork.cpp
@@ -182,6 +182,11 @@ void AT_CellularNetwork::attach(Callback<void(nsapi_event_t, intptr_t)> status_c
 
 nsapi_error_t AT_CellularNetwork::set_registration_urc(RegistrationType type, bool urc_on)
 {
+    tr_info("WTG set_registration_urc");
+    _at.cmd_start("AT+CGDCONT=1,\"IP\",\"XXX.com\"");
+    _at.cmd_stop_read_resp();
+
     int index = (int)type;
     MBED_ASSERT(index >= 0 && index < C_MAX);
 
@@ -223,7 +228,7 @@ nsapi_error_t AT_CellularNetwork::set_registration(const char *plmn)
         if (_op_act != RAT_UNKNOWN) {
             return _at.at_cmd_discard("+COPS", "=1,2,", "%s%d", plmn, _op_act);
         } else {
-            return _at.at_cmd_discard("+COPS", "=1,2", "%s", plmn);
+            return _at.at_cmd_discard("+COPS", "=1,2,", "%s", plmn);
         }
     }
 }
```

Logs 569 and 570 are using the following patch in order to enable the CREG registration, to fix 2.

```diff
diff --git a/features/cellular/framework/device/CellularStateMachine.cpp b/features/cellular/framework/device/CellularStateMachine.cpp
index 08b77977..633a5b20 100644
--- a/features/cellular/framework/device/CellularStateMachine.cpp
+++ b/features/cellular/framework/device/CellularStateMachine.cpp
@@ -708,8 +708,12 @@ void CellularStateMachine::cellular_event_changed(nsapi_event_t ev, intptr_t ptr
         // expect packet data so only these states are valid
         CellularNetwork::registration_params_t reg_params;
         nsapi_error_t err = _network.get_registration_params(reg_params);
+        tr_info("TEST: err %d, type %d", err, reg_params._type);
 
-        if (err == NSAPI_ERROR_OK && (reg_params._type == CellularNetwork::C_EREG || reg_params._type == CellularNetwork::C_GREG)) {
+        if (err == NSAPI_ERROR_OK && (reg_params._type == CellularNetwork::C_EREG
+                || reg_params._type == CellularNetwork::C_GREG
+                || reg_params._type == CellularNetwork::C_REG
+                )) {
             if ((data->status_data == CellularNetwork::RegisteredHomeNetwork ||
                     data->status_data == CellularNetwork::RegisteredRoaming) && data->error == NSAPI_ERROR_OK) {
                 _queue.cancel(_event_id);
```

#### Target(s) affected by this defect ?

All targets using the u-blox SARA-R412M (and perhaps other u-blox modules as well, especially the R4x series).

#### Toolchain(s) (name and version) displaying this defect ?

GCC_ARM

#### What version of Mbed-os are you using (tag or sha) ?

5.15.0
I didn't see significant changes in cellular in the release notes up to 5.15.3 except #12088 , which I manually patched in my 5.15.0 code.

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

```
$ mbed --version
1.10.0
$ arm-none-eabi-gcc --version
arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 9-2019-q4-major) 9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]
```

#### How is this defect reproduced ? 

- Do a cellular connection with the u-blox SARA-R412M and connect to a LTE Cat M1 network with a custom APN.
- Have APN configured using `CellularContext::set_credentials` and disable APN lookup.

AriParkkila, 2020-05-25:
@jdbruijn Thanks for the good logs.

Can you try if #13013 fixes the CREG problem. I can't reproduce this here, but according to the logs, CGREG/CEREG registration status was dismissed when received after CREG.

`CGDCONT` failure seems DNS related... Could you check with echo send/recv to see PDN is OK before DNS query?

jdbruijn, 2020-05-25:
> Can you try if #13013 fixes the CREG problem. I can't reproduce this here, but according to the logs, CGREG/CEREG registration status was dismissed when received after CREG.

Yes, I'll try this today and post back results.

> CGDCONT failure seems DNS related... Could you check with echo send/recv to see PDN is OK before DNS query?

I'm taking you mean to test this without the `CGDCONT` change I've made, so creating a socket (UDP) and sending a message to an IP, thus omitting the DNS query. Is that correct? If so, I'll check that too!

jdbruijn, 2020-05-25:
> Can you try if #13013 fixes the CREG problem. I can't reproduce this here, but according to the logs, CGREG/CEREG registration status was dismissed when received after CREG.

I've just tested with #13013 and that working just fine, without my patch for CREG obvs! Thanks for that patch! I'll manually apply that patch in my older OS version until I update to a newer release.

I'll check the DNS issue later today!

jdbruijn, 2020-05-26:
@AriParkkila It took a bit longer, but I've just tested echo without the `CGDCONT` change. Unfortunately, it didn't send the message although the device UDP socket thinks it has.

Here is my test code, and at our server I have netcat running (`nc -lu 1234`). I've again redacted the IP and port obvs.

```c++
        UDPSocket socket(network_connection.Interface());
        SocketAddress addr("x.x.x.x", 1234);

        char out_buffer[] = "time\n";
        nsapi_size_or_error_t e = socket.sendto(addr, out_buffer, sizeof(out_buffer));

        printf("Test message sent (%d)\n\n\n", e);
```

## Output without `CGDCONT`

This is the output without the `CGDCONT` (only the last section). Note that this the send returned 6, the size of the massage, but I didn't receive the message at the server.

```
[2020-05-26 15:29:43.073875] [INFO][CELL]: Found PDP context 1
[2020-05-26 15:29:43.077911] [INFO][CELL]: CellularContext PPP connect
[2020-05-26 15:29:43.081815] [INFO][CELL]: AT TX ( 9): ATD*99***
[2020-05-26 15:29:43.084871] [INFO][CELL]: AT TX ( 1): 1
[2020-05-26 15:29:43.089938] [INFO][CELL]: AT TX ( 1): #
[2020-05-26 15:29:43.091776] [INFO][CELL]: AT TX ( 1): <cr>
[2020-05-26 15:29:43.101924] [INFO][CELL]: AT RX (10): <cr><ln>CONNECT 
[2020-05-26 15:29:43.106925] [DBG ][CELL]: ppp_status_cb: event 0, ptr 3
[2020-05-26 15:29:43.110966] [DBG ][CELL]: callback: 0, ptr: 3
[2020-05-26 15:29:43.120450] [DBG ][CELL]: process_oob, filehandle is not usable, return...
[2020-05-26 15:29:49.729811] [DBG ][CELL]: ppp_status_cb: event 0, ptr 1
[2020-05-26 15:29:49.733903] [DBG ][CELL]: callback: 0, ptr: 1
[2020-05-26 15:29:49.737813] Connection status: 3, cellular status: 38
[2020-05-26 15:29:49.811918] Global IP received but waiting for local IP (30816 ms)
[2020-05-26 15:29:49.914868] Connected as X.X.X.X in 30920 ms
[2020-05-26 15:29:49.918824] [INFO][CELL]: CellularContext set blocking 1
[2020-05-26 15:29:49.920659] Connected to network!
[2020-05-26 15:29:49.921730]   RSSI: -51
[2020-05-26 15:29:49.923761] Test message sent (6)
[2020-05-26 15:29:49.923807] 
[2020-05-26 15:29:49.923821] 
[2020-05-26 15:31:29.487936] NSAPI address error! -3009
[2020-05-26 15:31:29.490894] Resolved IP address of platform ''
[2020-05-26 15:31:29.495957] [INFO][TLSW]: Starting TLS handshake with (null)
[2020-05-26 15:31:29.500989] [DBG ][TLSW]: mbedtls_ssl_setup()
[2020-05-26 15:31:29.506009] [ERR ][TLSW]: Socket send error -3006
[2020-05-26 15:31:29.517784] [ERR ][TLSW]: mbedtls_ssl_handshake() failed: -0x0bbe (-3006): UNKNOWN ERROR CODE (0B80) : ENTROPY - No more sources can be added
[2020-05-26 15:31:29.521902] Error: dtls.connect() returned -3011
```

## Output with `CGDCONT`

This is the output **with** the `CGDCONT` (only the last section). Note that this the send returned 6, again the size of the massage, and the message **is received** at the server.

```
[2020-05-26 15:28:21.469884] [INFO][CELL]: Found PDP context 1
[2020-05-26 15:28:21.474850] [INFO][CELL]: CellularContext PPP connect
[2020-05-26 15:28:21.478773] [INFO][CELL]: AT TX ( 9): ATD*99***
[2020-05-26 15:28:21.481818] [INFO][CELL]: AT TX ( 1): 1
[2020-05-26 15:28:21.485430] [INFO][CELL]: AT TX ( 1): #
[2020-05-26 15:28:21.488856] [INFO][CELL]: AT TX ( 1): <cr>
[2020-05-26 15:28:21.498049] [INFO][CELL]: AT RX ( 9): <cr><ln>CONNECT
[2020-05-26 15:28:21.505107] [DBG ][CELL]: ppp_status_cb: event 0, ptr 3
[2020-05-26 15:28:21.507034] [DBG ][CELL]: callback: 0, ptr: 3
[2020-05-26 15:28:21.514065] [DBG ][CELL]: process_oob, filehandle is not usable, return...
[2020-05-26 15:28:22.530705] [DBG ][CELL]: ppp_status_cb: event 0, ptr 1
[2020-05-26 15:28:22.534809] [DBG ][CELL]: callback: 0, ptr: 1
[2020-05-26 15:28:22.538777] Connection status: 3, cellular status: 38
[2020-05-26 15:28:22.613879] Global IP received but waiting for local IP (31715 ms)
[2020-05-26 15:28:22.715857] Connected as x.x.x.x in 31819 ms
[2020-05-26 15:28:22.719847] [INFO][CELL]: CellularContext set blocking 1
[2020-05-26 15:28:22.721860] Connected to network!
[2020-05-26 15:28:22.722862]   RSSI: -51
[2020-05-26 15:28:22.725878] Test message sent (6)
[2020-05-26 15:28:22.725928] 
[2020-05-26 15:28:22.725941] 
[2020-05-26 15:28:23.050074] Resolved IP address of platform 'x.x.x.x'
[2020-05-26 15:28:23.055019] [INFO][TLSW]: Starting TLS handshake with (null)
[2020-05-26 15:28:23.059866] [DBG ][TLSW]: mbedtls_ssl_setup()
[2020-05-26 15:28:25.607752] [INFO][TLSW]: TLS connection to (null) established
[2020-05-26 15:28:25.612156] [DBG ][TLSW]: Server certificate:
...
[2020-05-26 15:28:25.658797] [INFO][TLSW]: Certificate verification passed
[2020-05-26 15:28:25.661861] DTLS Open finished, Connected now!
```

AriParkkila, 2020-05-27:
@jdbruijn Thanks for testing. Obviously, the modem should not tell it sent something it didn't... @ARMmbed/team-ublox could know more about that?

jdbruijn, 2020-06-03:
@AriParkkila Any update on the outstanding issue as described in https://github.com/ARMmbed/mbed-os/issues/13000#issuecomment-634033722?

AriParkkila, 2020-06-04:
> it didn't send the message although the device UDP socket thinks it has

@jdbruijn @ARMmbed/team-ublox can probably answer what might wrong on the modem side.

Culprit could be something like a conflict between the default initial context and your PDN, so you could still check `AT+CGDCONT?` that APN of PDP with ID 1 is correct (in the logs X/XXX.com) before network is attached `+CGATT: 1`.

jdbruijn, 2020-06-04:
> Culprit could be something like a conflict between the default initial context and your PDN, so you could still check `AT+CGDCONT?` that APN of PDP with ID 1 is correct (in the logs X/XXX.com) before network is attached `+CGATT: 1`.

Thanks for the suggestion. I've tried this by adding the `AT+CGDCONT?` to the beginning of `AT_CellularNetwork::get_signal_quality` since I noticed that the signal quality is read before `+CGATT: 1` in the logs.

The default for the module is all zeroes ([u-blox AT manual](https://www.u-blox.com/en/docs/UBX-17003787#%5B%7B%22num%22%3A1822%2C%22gen%22%3A0%7D%2C%7B%22name%22%3A%22XYZ%22%7D%2C59.527%2C678.177%2Cnull%5D)) so looks something like below, but not all is read due to AT buffer size.
During the first two thirds or so of the logs, the response of `AT+CGDCONT?` is the default. After network registration, but before `+CGATT: 1`, the response shows the correct APN (redacted of course) and its assigned IP (see 2nd part of the log below).
So it seems I do see the correct APN before network is attached. If there is any more I could test please let me know!

```
[2020-06-04 09:16:31.418314] [INFO][CELL]: AT ERR (32): +CGDCONT: 1,"IPV4V6","","0.0.0.0
[2020-06-04 09:16:31.423924] [INFO][CELL]: AT RX (32): .0.0.0.0.0.0.0.0.0.0.0.0",0,0,0,
[2020-06-04 09:16:31.428149] [ERR ][CELL]: AT overflow

...

[2020-06-04 09:16:39.287177] [INFO][CELL]: AT ERR (32): +CGDCONT: 1,"IP","example.co
[2020-06-04 09:16:39.295146] [INFO][CELL]: AT RX (32): m","1.2.3.4",0,0,0,0<cr><ln><cr><ln>OK<cr>
[2020-06-04 09:16:39.298115] [INFO][CELL]: AT RX ( 1): <ln>
```

@AriParkkila I've noted that the default is an IPV4V6 address and the configured/received address is IPV4, so could that have anything to do with this issue? Do I need change the default to all zeroes, but with IPV4?

<details>
<summary>Full log</summary>

```
[2020-06-04 09:16:08.513350] [INFO][CELL]: CellularContext set blocking 0
[2020-06-04 09:16:08.528134] [INFO][CELL]: CellularContext set blocking 0
[2020-06-04 09:16:08.529230] [INFO][CELL]: CellularContext connect: CellularContext connect
[2020-06-04 09:16:08.539153]
[2020-06-04 09:16:08.542301] Connection status: 2, cellular status: 0
[2020-06-04 09:16:08.542301] [DBG ][CELL]: callback: 4106, err: 0, data: 4096
[2020-06-04 09:16:08.561442] Unknown cellular status
[2020-06-04 09:16:08.569459] Connection status: 2, cellular status: 0
[2020-06-04 09:16:08.579424] [INFO][CELL]: Start connecting (timeout 1000 ms)
[2020-06-04 09:16:08.582425] [INFO][CELL]: AT TX ( 2): AT
[2020-06-04 09:16:08.591487] [INFO][CELL]: AT TX ( 1): <cr>
[2020-06-04 09:16:09.584441] [DBG ][CELL]: AT error -3012
[2020-06-04 09:16:09.588261] [INFO][CELL]: AT TX ( 2): AT
[2020-06-04 09:16:09.591439] [INFO][CELL]: AT TX ( 1): <cr>
[2020-06-04 09:16:10.589238] [DBG ][CELL]: AT error -3012
[2020-06-04 09:16:10.592189] [DBG ][CELL]: Init => Power
[2020-06-04 09:16:10.597199] [DBG ][CELL]: callback: 4106, err: 0, data: 4096
[2020-06-04 09:16:10.599171] Unknown cellular status
[2020-06-04 09:16:10.603248] Connection status: 2, cellular status: 0
[2020-06-04 09:16:10.608188] [INFO][CELL]: Modem power ON (timeout 1000 ms)
[2020-06-04 09:16:10.631269] [DBG ][CELL]: Power => Device ready
[2020-06-04 09:16:10.637227] [DBG ][CELL]: callback: 4106, err: 0, data: 4096
[2020-06-04 09:16:10.639202] Unknown cellular status
[2020-06-04 09:16:10.642200] Connection status: 2, cellular status: 0
[2020-06-04 09:16:10.648213] [DBG ][CELL]: Device was not ready, calling soft_power_on()
[2020-06-04 09:16:10.850281] [DBG ][CELL]: AT flush
[2020-06-04 09:16:10.854231] [INFO][CELL]: AT RX ( 1): <00>
[2020-06-04 09:16:10.858249] [INFO][CELL]: AT TX ( 4): ATE0
[2020-06-04 09:16:10.861248] [INFO][CELL]: AT TX ( 1): <cr>
[2020-06-04 09:16:11.858572] [DBG ][CELL]: AT error -3012
[2020-06-04 09:16:11.863334] [DBG ][CELL]: Wait 100ms to init modem
[2020-06-04 09:16:11.965625] [DBG ][CELL]: AT flush
[2020-06-04 09:16:11.968502] [INFO][CELL]: AT TX ( 4): ATE0
[2020-06-04 09:16:11.973879] [INFO][CELL]: AT TX ( 1): <cr>
[2020-06-04 09:16:12.970577] [DBG ][CELL]: AT error -3012
[2020-06-04 09:16:12.973282] [DBG ][CELL]: Wait 100ms to init modem
[2020-06-04 09:16:13.075361] [DBG ][CELL]: AT flush
[2020-06-04 09:16:13.079551] [INFO][CELL]: AT TX ( 4): ATE0
[2020-06-04 09:16:13.083446] [INFO][CELL]: AT TX ( 1): <cr>
[2020-06-04 09:16:14.080638] [DBG ][CELL]: AT error -3012
[2020-06-04 09:16:14.084574] [DBG ][CELL]: Wait 100ms to init modem
[2020-06-04 09:16:14.188310] [DBG ][CELL]: Device ready: retry 0/10
[2020-06-04 09:16:14.196315] [DBG ][CELL]: callback: CellularStateRetryEvent, err: 0, data: 4096, retrycount: 0
[2020-06-04 09:16:14.198711] Unknown cellular status
[... 299 lines cut ...]
[2020-06-04 09:16:39.400227] [INFO][CELL]: AT RX ( 1): <ln>
[2020-06-04 09:16:39.404198] [INFO][CELL]: Found PDP context 1
[2020-06-04 09:16:39.408089] [INFO][CELL]: CellularContext PPP connect
[2020-06-04 09:16:39.412178] [INFO][CELL]: AT TX ( 9): ATD*99***
[2020-06-04 09:16:39.416225] [INFO][CELL]: AT TX ( 1): 1
[2020-06-04 09:16:39.419169] [INFO][CELL]: AT TX ( 1): #
[2020-06-04 09:16:39.423085] [INFO][CELL]: AT TX ( 1): <cr>
[2020-06-04 09:16:39.431203] [INFO][CELL]: AT RX ( 5): <cr><ln>CON
[2020-06-04 09:16:39.437196] [INFO][CELL]: AT RX (16): NECT 150000000<cr><ln>
[2020-06-04 09:16:39.443190] [DBG ][CELL]: ppp_status_cb: event 0, ptr 3
[2020-06-04 09:16:39.446210] [DBG ][CELL]: callback: 0, ptr: 3
[... cut ...]
[2020-06-04 09:16:46.341525] Connection status: 3, cellular status: 38
[2020-06-04 09:16:46.375466] Global IP received but waiting for local IP (37812 ms)
[2020-06-04 09:16:46.478566] Connected as 1.2.3.4 in 37916 ms
[2020-06-04 09:16:46.483564] [INFO][CELL]: CellularContext set blocking 1
[2020-06-04 09:16:46.485603] Connected to network!
[2020-06-04 09:16:46.486485]   RSSI: -67
[2020-06-04 09:16:46.488527] Test message sent (6)
[2020-06-04 09:16:46.488629]
[2020-06-04 09:16:46.488672]
```
</details>

AriParkkila, 2020-06-04:
@jdbruijn It might be possible that the default context with an empty APN ("") is activated here. To prevent that you might try with JSON configurations `cellular.clear-on-connect` and `nsapi.default-cellular-apn` to define APN early enough.

jdbruijn, 2020-06-04:
I've tried both, but makes no difference. I was already setting the credentials in my code before connecting. Also, `AT_CellularNetwork::clear` isn't even called according to my logs, which is the only place where `MBED_CONF_NSAPI_DEFAULT_CELLULAR_APN` is really used. @AriParkkila should that function be called?
I've overridden the `CellularDevice::get_default_instance` to set my own cellular device drivers so `CellularInterface::set_default_parameters`, the other place where `MBED_CONF_NSAPI_DEFAULT_CELLULAR_APN` is used, isn't called but I set those credentials as shown below.

```cpp
  _connection = mbed::CellularContext::get_default_instance();
  _interface = _connection;
  _connection->set_credentials("example.com", 0, 0);
```

Mostly for my personal reference: this is with the config defines below.
```cpp
#define MBED_CONF_NSAPI_DEFAULT_CELLULAR_APN "example.com"
#define MBED_CONF_CELLULAR_CLEAR_ON_CONNECT 1
```

jdbruijn, 2020-06-04:
Okay, this is quite strange. I noticed that the code inside the `MBED_CONF_CELLULAR_CLEAR_ON_CONNECT` check `#ifdef` wasn't reached. I don't know why yet because it surely was configured in _mbed_config.h_, but when I manually removed it, something changed. I now as able to send a UDP packet to my server with only the clear on connect change, as I already configured the APN in the credentials call. In the logs I see that after the clear, it immediately sets the APN, so it indeed looks like something with the context although I don't quite understand it. In de end, it still sets the context previously so I don't get why it would run into the issue thinking it has sent data but not actually sent it. 

```
[2020-06-04 12:26:26.762406] [INFO][CELL]: SIM is ready
[2020-06-04 12:26:26.766990] [DBG ][CELL]: callback: 4097, err: 0, data: 0
[2020-06-04 12:26:26.771066] Connection status: 2, cellular status: 2
[2020-06-04 12:26:26.774100] [INFO][CELL]: SIM_READY
[2020-06-04 12:26:26.777005] [INFO][CELL]: CLEAR_ON_CONNECT
[2020-06-04 12:26:26.781097] [INFO][CELL]: AT_CellularNetwork::clear
[2020-06-04 12:26:26.786190] [INFO][CELL]: AT TX (11): AT+CGDCONT?
[2020-06-04 12:26:26.789017] [INFO][CELL]: AT TX ( 1): <cr>
[2020-06-04 12:26:26.793032] [INFO][CELL]: AT RX ( 7): <cr><ln>+CGDC
[2020-06-04 12:26:26.799097] [INFO][CELL]: AT RX (27): ONT: 1,"IP","example.co
[2020-06-04 12:26:26.805996] [INFO][CELL]: AT RX (28): m","0.0.0.0",0,0,0,0<cr><ln><cr><ln>OK<cr><ln>
[2020-06-04 12:26:26.811051] [DBG ][CELL]: Automatic network registration
```

And the change I had to do to work around the still unknown define issue.

```diff
diff --git a/features/cellular/framework/device/CellularStateMachine.cpp b/features/cellular/framework/device/CellularStateMachine.cpp
index 08b779773b..b82308a4b3 100644
--- a/features/cellular/framework/device/CellularStateMachine.cpp
+++ b/features/cellular/framework/device/CellularStateMachine.cpp
@@ -173,12 +173,14 @@ bool CellularStateMachine::open_sim()
     bool sim_ready = state == CellularDevice::SimStateReady;

     if (sim_ready) {
-#ifdef MBED_CONF_CELLULAR_CLEAR_ON_CONNECT
+      tr_info("SIM_READY");
+// #ifdef MBED_CONF_CELLULAR_CLEAR_ON_CONNECT
+      tr_info("CLEAR_ON_CONNECT");
         if (_cellularDevice.clear() != NSAPI_ERROR_OK) {
             tr_warning("CellularDevice clear failed");
             return false;
         }
-#endif
+// #endif
         _cb_data.error = _network.set_registration(_plmn);
         tr_debug("STM: set_registration: %d, plmn: %s", _cb_data.error, _plmn ? _plmn : "NULL");
         if (_cb_data.error) {
```

<details>
<summary>Full log</summary>
```
[2020-06-04 12:26:25.313017] [DBG ][CELL]: Wait 100ms to init modem
[2020-06-04 12:26:25.416963] [DBG ][CELL]: Device ready: retry 0/10
[2020-06-04 12:26:25.424954] [DBG ][CELL]: callback: CellularStateRetryEvent, err: 0, data: 4096, retrycount: 0
[2020-06-04 12:26:25.427030] Unknown cellular status
[2020-06-04 12:26:25.431003] Connection status: 2, cellular status: 0
[2020-06-04 12:26:25.435265] [INFO][CELL]: Continue after 1 seconds
[2020-06-04 12:26:26.436261] [DBG ][CELL]: callback: 4106, err: 0, data: 4096
[2020-06-04 12:26:26.438225] Unknown cellular status
[2020-06-04 12:26:26.440999] Connection status: 2, cellular status: 0
[2020-06-04 12:26:26.446983] [DBG ][CELL]: Device was not ready, calling soft_power_on()
[2020-06-04 12:26:26.651991] [WARN][CELL]: URC already added with prefix: +CGEV: NW DEACT
[2020-06-04 12:26:26.659044] [WARN][CELL]: URC already added with prefix: +CGEV: ME DEACT
[2020-06-04 12:26:26.663999] [WARN][CELL]: URC already added with prefix: +CGEV: NW PDN D
[2020-06-04 12:26:26.670056] [WARN][CELL]: URC already added with prefix: +CGEV: ME PDN D
[2020-06-04 12:26:26.673406] [DBG ][CELL]: AT flush
[2020-06-04 12:26:26.677083] [INFO][CELL]: AT TX ( 4): ATE0
[2020-06-04 12:26:26.681712] [INFO][CELL]: AT TX ( 1): <cr>
[2020-06-04 12:26:26.686045] [INFO][CELL]: AT RX (11): ATE0<cr><cr><ln>OK<cr><ln>
[2020-06-04 12:26:26.690233] [INFO][CELL]: AT TX ( 9): AT+CMEE=1
[2020-06-04 12:26:26.693061] [INFO][CELL]: AT TX ( 1): <cr>
[2020-06-04 12:26:26.698036] [INFO][CELL]: AT RX ( 6): <cr><ln>OK<cr><ln>
[2020-06-04 12:26:26.702422] [INFO][CELL]: AT TX ( 9): AT+CFUN=1
[2020-06-04 12:26:26.705817] [INFO][CELL]: AT TX ( 1): <cr>
[2020-06-04 12:26:26.710182] [INFO][CELL]: AT RX ( 6): <cr><ln>OK<cr><ln>
[... 273 lines cut ...]
[2020-06-04 12:26:50.383849] [INFO][CELL]: AT TX ( 9): ATD*99***
[2020-06-04 12:26:50.386751] [INFO][CELL]: AT TX ( 1): 1
[2020-06-04 12:26:50.390879] [INFO][CELL]: AT TX ( 1): #
[2020-06-04 12:26:50.393745] [INFO][CELL]: AT TX ( 1): <cr>
[2020-06-04 12:26:50.402874] [INFO][CELL]: AT RX ( 3): <cr><ln>C
[2020-06-04 12:26:50.407740] [INFO][CELL]: AT RX (18): ONNECT 150000000<cr><ln>
[2020-06-04 12:26:50.413065] [DBG ][CELL]: ppp_status_cb: event 0, ptr 3
[2020-06-04 12:26:50.416726] [DBG ][CELL]: callback: 0, ptr: 3
[2020-06-04 12:26:50.423304] [DBG ][CELL]: process_oob, filehandle is not usable, return...
[2020-06-04 12:26:51.558997] [DBG ][CELL]: ppp_status_cb: event 0, ptr 1
[2020-06-04 12:26:51.561999] [DBG ][CELL]: callback: 0, ptr: 1
[2020-06-04 12:26:51.565817] Connection status: 3, cellular status: 38
[2020-06-04 12:26:51.587753] Global IP received but waiting for local IP (31912 ms)
[2020-06-04 12:26:51.690005] Connected as 1.2.3.4 in 32016 ms
[2020-06-04 12:26:51.694874] [INFO][CELL]: CellularContext set blocking 1
[2020-06-04 12:26:51.696744] Connected to network!
[2020-06-04 12:26:51.697750]   RSSI: -51
[2020-06-04 12:26:51.699914] Test message sent (6)
[2020-06-04 12:26:51.700002]
[2020-06-04 12:26:51.700030]
[2020-06-04 12:26:52.003029] Resolved IP address of platform 'x.x.x.x'
[2020-06-04 12:26:52.007904] [INFO][TLSW]: Starting TLS handshake with (null)
```

</details>

AriParkkila, 2020-06-04:
@jdbruijn That sounds weird, build cleanup could help... at least to get rid of any excessive mbed_config.h files.

Good it's working now. Effectively `clear()` should do pretty much the same as your workaround, i.e. set PDP ID 1 APN. Another possible fix could be to use the next but the initial context, but I don't know if that's in any way better.

jdbruijn, 2020-06-04:
@AriParkkila Okay, thanks for explaining. So do you recommend me just keep using the `MBED_CONF_CELLULAR_CLEAR_ON_CONNECT` (`cellular.clear-on-connect`) configuration?

Also, do you have any idea why this issue appears on the SARA-R412M/R410M and not on the SARA-G350?

As far as my issues in this ticket: both are now resolved as far as I consider. The issue with `MBED_CONF_CELLULAR_CLEAR_ON_CONNECT` looks to be on my side as its just a preprocessor macro so should work just fine. There still is the issue with the modem thinking it has sent something while I don't receive anything so as far as I'm considered we can either keep this issue open to track that or you can track it internally/create a different issue. In case of the latter, this issue may be closed.

AriParkkila, 2020-06-04:
@jdbruijn CID=1 probably needs clear-flag, but could probably also try to use some other CID. It should work without clear, so it could be a less "obstructive" in some cases, see e.g. [context-id-offset](https://github.com/AriParkkila/mbed-os/commit/6c64f9b3ec62896c5754153fd71471c6abfc1b69)

The difference between R412M/R410M and G350 could be that one is not saving the APN values in NVM when switching off the modem, or simply because G350 could be in GPRS network without combined attached and initial context... It's hard to say for sure as I'm not a specialist with these modems.

The issue with the modem thinking it has sent something and actually didn't, sounds like a modem bug.