ARMmbed/mbed-os #13004: Mbed OS example application doesn't run for MUSCA A1 target

jainvikas8, 2020-05-21:
### Description of defect

Mbed OS example application not running for target - ARM_MUSCA_A1
A simple https://github.com/ARMmbed/mbed-os-example-blinky example results in error

--- Error ---
`[INF] Starting bootloader [ERR] Unable to find bootable image`

#### Target(s) affected by this defect ?
ARM_MUSCA_A1

#### Toolchain(s) (name and version) displaying this defect ?
`mbed compile -t GCC_ARM -m ARM_MUSCA_A1_NS`
`mbed compile -t ARMC6 -m ARM_MUSCA_A1_NS`

#### What version of Mbed-os are you using (tag or sha) ?

https://github.com/ARMmbed/mbed-os/commit/9a8c9e2c297f64c805698d72cd541ff3cd7fe538

#### How is this defect reproduced ? 

Use https://github.com/ARMmbed/mbed-os/commit/9a8c9e2c297f64c805698d72cd541ff3cd7fe538 version of mbed-os with https://github.com/ARMmbed/mbed-os-example-blinky. Build and flash the binary to MUSCA A1 board.

jainvikas8, 2020-05-22:
Effects - https://github.com/ARMmbed/mbed-os-example-crash-reporting/issues/46 and https://github.com/ARMmbed/mbed-os-example-thread-statistics/issues/57

0xc0170, 2020-05-28:
I updated the release version.

adbridge, 2020-06-15:
@jainvikas8 it looks like you removed the tools versions section from the issue template ('What version(s) of tools are you using. List all that apply (E.g. mbed-cli)') . This is preventing the issue from being mirrored! Please do not remove sections! Please re-instate it and fill it in. See https://github.com/ARMmbed/mbed-os/issues/13106 as an example.

MarceloSalazar, 2020-06-22:
@jainvikas8 
Note the Musca A1 is not considered supported in Mbed OS 6, hence we plan to remove it - https://github.com/ARMmbed/mbed-os/pull/13165

MarceloSalazar, 2020-06-22:
Closing as https://github.com/ARMmbed/mbed-os/issues/13106 is merged and Musca_A1 isn't supported.

closed 2020-06-22 by MarceloSalazar