ARMmbed/mbed-os #13005: AnalogOut Works in Online Compiler, But Not When Using MBED CLI

davidantaki, 2020-05-21:
### Description of defect

When writing to a DAC1/1 on pin PA_4 on a NUCLEO L432KC using a for loop that incrementally increases the percent of the Vref to be outputted, the DAC output is seemingly left floating as the voltage output jumps around (observed with multimeter).

However, this same code works when using the online compiler and also works when there is no for loop. When I simply write a float value to the DAC (with no for loop) the DAC outputs the expected voltage.

#### Target(s) affected by this defect ?
NUCLEO_L432KC

#### Toolchain(s) (name and version) displaying this defect ?
arm-none-eabi-g++ 6.3.1 20170620 (release) 
GCC_ARM 

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-5.15.3

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
mbed-cli 1.10.2
VSCode 1.45.1
GNU bash (Git Bash), version 4.4.23(1)-release (x86_64-pc-msys)

#### How is this defect reproduced ? 
I create a new Mbed project using mbed-cli with:
`mbed new testProj`

I set default toolchain with:
`mbed toolchain GCC_ARM`

I set default target with:
`mbed target NUCLEO_L432KC`

I make a main.cpp with the following code:
```
#include "mbed.h"

int main()
{
    AnalogOut control(PA_4); 
    Serial pc(USBTX, USBRX, 9600);

    while (1)
    {
        for (float i = 0; i < 1.0f; i += 0.01f)
        {
            control.write(i);
            pc.printf("DAC Float Val: %f\n", i);
            wait(5);
        }
    }
}
```

I compile in terminal with:
`mbed compile -v -f`

Nucleo board is flashed and the for loop is running since it's printing out to Serial. However, when observing PA_4 (A3) with a multimeter, it outputs ~0.9V and jumps around instead of slowly increasing.

However, this same code works when using the online compiler. Further, the following works just fine when using mbed-cli:
```
#include "mbed.h"

int main()
{
    AnalogOut control(PA_4);
    Serial pc(USBTX, USBRX, 9600);

    while (1)
    {
             control.write(.3);
    }
}

```

jeromecoutant, 2020-07-02:
Thx @davidantaki for issue report.

I think I understand the issue now!
- online compiler should use debug profile. Then deepsleep is not enabled.
- with mbed cli, deepsleep is then enabled, and DAC feature is then reset during wait period...

@0xc0170 
you could set device st label

closed 2020-07-03 by 0xc0170

davidantaki, 2020-07-05:
Thanks for the info/fix, much appreciated. @jeromecoutant

PR #13221 (2020-07-03): STM32 ANALOGOUT and DEEPSLEEP
### Summary of changes 

Fix #13005 

@davidantaki

When deepsleep is enabled, DAC is reset during wait period.

Change is then to disable deepsleep during analogout

#### Impact of changes 

#### Migration actions required 

### Documentation 

----------------------------------------------------------------------------------------------------------------
### Pull request type