ARMmbed/mbed-os #13010: BLE Security passkey error

prostoAlexandr, 2020-05-23:
### Description of defect

Hello! I have a problem with a ble security using the nrf52832 target. I tried different variants of configuration and no one helped. I asked similar question on the mbed forum, but nobody answered. I tried to build my program with vsCode+platformio, which uses the mbed version 5.14. Also I used mbed online compiler. The problem is still the same. It is as follows: when I trying to use ble security with password, the pairing fails after I send the password to the controller. It was chosen "IO_CAPS_DISPLAY_ONLY" parameter and I use "passkeyDisplay" callback to output generated password to the COM-port. I used an example mbed-os-example-ble/BLE_SM/ from mbed ble examples. The code is below:

    #include <mbed_config.h>
    #include <events/mbed_events.h>
    #include <mbed.h>
    #include "ble/BLE.h"
    #include "ble/Gap.h"
    #include "ble/SecurityManager.h"
    #include "pretty_printer.h"
    #include "ConnectionParameters.h"

    #define MBED_CONF_APP_FILESYSTEM_SUPPORT 1

    #if MBED_CONF_APP_FILESYSTEM_SUPPORT
    #include "LittleFileSystem.h"
    #include "HeapBlockDevice.h"
    #endif 

    static const char DEVICE_NAME[] = "SM_device";
    static ble::address_t peer_address;

    class SMDevice : private mbed::NonCopyable<SMDevice>,
                    public SecurityManager::EventHandler,
                    public ble::Gap::EventHandler
    {
    public:
        SMDevice(BLE &ble, events::EventQueue &event_queue, ble::address_t &peer_address) :
            _led1(P0_26, 0),
            _ble(ble),
            _event_queue(event_queue),
            _peer_address(peer_address),
            _handle(0),
            _is_connecting(false) { };

        virtual ~SMDevice()
        {
            if (_ble.hasInitialized()) {
                _ble.shutdown();
            }
        };

        /** Start BLE interface initialisation */
        void run()
        {
            ble_error_t error;
            _event_queue.call_every(500, this, &SMDevice::blink);

            if (_ble.hasInitialized()) {
                printf("Ble instance already initialised.\r\n");
                return;
            }

            _ble.onEventsToProcess(
                makeFunctionPointer(this, &SMDevice::schedule_ble_events)
            );

            _ble.gap().setEventHandler(this);

            error = _ble.init(this, &SMDevice::on_init_complete);
            if (error) {
                printf("Error returned by BLE::init.\r\n");
                return;
            }

            _event_queue.dispatch_forever();
        };

    private:
[... 204 lines cut ...]
At the end I tried to use static passkey, but it changed nothing.

I also tried to debug my application. The debugger via vsCode+Platformio+mbed doesn't work fine. It stops the program in wrong places sometimes. At the time when I send the passkey to the controller the program could fail because of stack overflow as far as I understand. It happens only during debugging. I tried to increase the thread size, but it didn't help.
Debugging with only vsCode+mbed-cli doesn't work too (described in https://github.com/ARMmbed/mbed-os-5-docs/issues/1279).
The mbed studio doesn't see my target, because it is not bult-in.
At the moment I don't see another possible way except trying to install some another IDE.
Maybe someone could help me to find the way how to debug my platform? Maybe I use API wrong? Maybe it is really a bug?

#### Target(s) affected by this defect ?
nrf52-DK(the most close to my nrf52832 based module E73-2G4M04S).

#### Toolchain(s) (name and version) displaying this defect ?
ST-link v2 (from the Nucleo-F401RE board).

#### What version of Mbed-os are you using (tag or sha) ?

5.14 (platformio)
default version for mbed online compiler

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
vsCode 1.39.2
Platformio 1.10.0

#### How is this defect reproduced ? 
Very easy. It doesn't work properly anyway.

prostoAlexandr, 2020-05-24:
Hello! I provided the first comment with the platformio mbed version framework-mbed-6.51401.200402. Is it OK? If not, then where I could find the SHA or release tag of the my os version?

prostoAlexandr, 2020-05-25:
Ok, it looks like not good. I changed the os version to n/a for a while because I can't find how to know it.

pan-, 2020-10-05:
@prostoAlexandr What application do you use to pair to the device ? On which OS ?

prostoAlexandr, 2020-10-06:
@pan- It was my laptop on windows 10, as well as my cell phone powered by android 10. The result was the same. The application is a default system bluetooth pairing service.

paul-szczepanek-arm, 2021-02-05:
Thank you for bringing this to our attention. The passkey is stored as a single 6 byte number in little endian. You basically have to display it backwards. Sorry about that.

I will fix the callback to reverse the passkey. This will have to be done through a new callback and deprecating the current one so as not to break it for people who already use it as is.

I will report back when I have a PR.

paul-szczepanek-arm, 2021-02-08:
https://github.com/ARMmbed/mbed-os/pull/14239

paul-szczepanek-arm, 2021-02-08:
So it's simpler than that, the string is reversed and you just need to set an option now and it will apply the fix.

The option is there to avoid any accidental breakage for people who use it as is.

closed 2021-02-10 by ciarmcom