ARMmbed/mbed-os #13012: Some sort of limit on number of files open with SDFileSystem

roryhand, 2020-05-23:
### Description of defect

  When attempting to open more than 13 .wav files at once using the sdfilesystem in Mbed OS-2, the board starts returning errors, and so does not open the files.  I previously had this issue with an LPC1768 which prevented me from continuing with an application I was working on (a polyphonic sound playing device for model railroad locomotives).  From performing various tests, (including with 1KB text files), I concluded that this was due to the limited amount of RAM available on this chip (32KB) and the large overhead used by the FATFS when opening virtually any type of file.

I switched to using the NUCLEO-F429ZI, and I have exactly the same issue.  This is despite the chip having 256KB of SRAM.  I was surprised, as previously I had performed a test using an LPC55S69EVK board, in MCUXpresso using the appropriate MCUXpresso SDK.  (There is an example using FATFS with sd cards).

Using exactly the same .wav files I had been able to open 22 different files at once without any memory issues in MCUXpresso using their sdk; the NXP chip in question has 320KB of RAM, and only 10% was used.

So there seems to me that there is some sort of limit to the number of files within SDFileSystem that is causing issues, even on devices with lots of RAM.
  

#### Target(s) affected by this defect ?
NUCLEO-F429ZI (possibly others)

#### Toolchain(s) (name and version) displaying this defect ?

#### What version of Mbed-os are you using (tag or sha) ?

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

#### How is this defect reproduced ?

0xc0170, 2020-05-25:
Can you provide a reference to filesystem library - to avoid confusion. As you state it's for Mbed 2, so its not part of this repository.

kjbracey-arm, 2020-05-25:
The retargetting code, which is probably still fairly common, has a fixed limit, due to a static filehandle array. 13 files would correspond to that being 16, minus stdin/stdout/stderr.

Looking in the code here https://github.com/ARMmbed/mbed-os/blob/9a8c9e2c297f64c805698d72cd541ff3cd7fe538/platform/source/mbed_retarget.cpp#L113 the macro `RETARGET_OPEN_MAX` probably needs to be increased.

There's no exposed config mechanism in Mbed OS 6 to raise that limit - you'd have to edit the C++ file.

Whether that's sufficient would depend on the C library. Not sure if they have their own internal limits. Probably not.

roryhand, 2020-05-26:
Hi @kjbracey-arm, thanks for pointing this out.  Is there a way I can change this in the code within the mbed compiler environment?  When I click on the mbed-os library within any given project, I cannot view any documentation directly, or any code, so I would have no way of changing this RETARGET_OPEN_MAX macro.

roryhand, 2020-05-26:
Hi @0xc0170 , the filesystem in question is the one found here:  https://os.mbed.com/users/mbed_official/code/SDFileSystem/

I am not sure where I would find this on github or similar.  Is there a way I can raise this issue on the appropriate github repository (I am a total newbie to github, so not very clear how it all works).  So that this issue isn't raised on this incorrect repository.  Cheers.

0xc0170, 2020-05-27:
@roryhand you should be able to set macro, similar to `-DRETARGET_OPEN_MAX=16`

adbridge, 2020-06-23:
@roryhand as there has been no further update in the last 25 days we will close this. Please re-open if you still have an issue. Thanks

closed 2020-06-23 by adbridge