ARMmbed/mbed-os #13018: Backhaul interface isn't reseting when conection is lost

AlefeFelipe, 2020-05-26:
### Description of defect

Backhaul interface isn't resetting after a connection loss. The log is attached. We can see in the borderrouter_ws.c code that there is a callback that indicates that the backhaul lost connection but nothing is done after that. We waited 2 hours and nothing, only got back after manual reset. Wisun network are supposed to be self healing.

#### Target(s) affected by this defect ?
NUCLEO_WB55RG

#### Toolchain(s) (name and version) displaying this defect ?
GCC_ARM

#### What version of Mbed-os are you using (tag or sha) ?

5.15.3

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
mbed-cli 1.10.1

#### How is this defect reproduced ? 
Start a application with border router, and after the Backhaul is up, wait for it get down

[debug.log](https://github.com/ARMmbed/mbed-os/files/4682329/debug.log)

0xc0170, 2020-05-27:
cc @artokin

mikter, 2020-05-28:
Backbone interface was shutdown by application Border router can't recover from that if the interfaces are not restarted and border router needs to be restarted as it has wrong interface information.

Interfaces should not be disconnected by application as it will remove the possibility for border router to function

AlefeFelipe, 2020-05-28:
Thanks for the fast reply, 

I undestood what you said and I agree, the backhaul is beeing disconnected by the application when the application see that the backhaul interface has no long internet connection. I'm using the reference from [nanostack-border-router](https://github.com/ARMmbed/nanostack-border-router). 

We saw that the reason for this behavior is that, in our case, the CellularInterface loses connection and triggers a callback from the application (`borderrouter_backhaul_phy_status_cb`) and this callback calls for an event that brings the backhaul down, and never up again:
```c
} else if (event->event_id == NR_BACKHAUL_INTERFACE_PHY_DOWN) {
    if (backhaul_interface_down() == 0) {
        tr_debug("Backhaul interface is down");
    }
}
```
So our question is more related to what to do next, should we put a `backhaul_interface_up()` or reset the CellularInterface, if you could help us here we can create a PR on [nanostack-border-router](https://github.com/ARMmbed/nanostack-border-router) for this.

mikaleppanen, 2020-05-29:
How the backhaul interface goes down? @AlefeFelipe is it the PPP that disconnects? If that is the case could you enable the PPP traces on /mbed-os/features/netsocket/ppp/mbed_lib.json and take a log file about PPP:

        "debug": {
            "help": "Enable debug traces for PPP",
            "value": false,                                                              <---------- change to true
            "macro_name": "PPP_DEBUG"
        },
        "enable-trace": {
            "help": "Enable PPP packet traces (traces every outbound and inbound packet)",
           "value": false                                                                 <--------- change to true
        },

Those will give more information what occurs.

AlefeFelipe, 2020-05-29:
Thanks, I'll do that. I'll be back to you once a have the log

mikaleppanen, 2020-09-10:
@AlefeFelipe is this valid anymore? Can we close this.

0xc0170, 2020-09-10:
I'll close. If still valid, it can be reopened with an update.

closed 2020-09-10 by 0xc0170