ARMmbed/mbed-os #13021: LED use in greentea tests ?

jeromecoutant, 2020-05-27:
### Description of defect

TESTS-MBED_DRIVERS-TICKER needs 2 LED to be defined:
https://github.com/ARMmbed/mbed-os/blob/master/TESTS/mbed_drivers/ticker/main.cpp#L42-L43

Is it really mandatory for an automatic test ?

@jamesbeyond 

#### Target(s) affected by this defect ?

All

#### Toolchain(s) (name and version) displaying this defect ?

NA

#### What version of Mbed-os are you using (tag or sha) ?

NA

Master branch mbed-os-99.99.99

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

NA

#### How is this defect reproduced ?

0xc0170, 2020-05-27:
I think we can remove these. Are they causing any issues ?

jeromecoutant, 2020-05-27:
Some boards have only 1 LED.
Goal was to remove workaround like "#define LED2 LED1"

kjbracey-arm, 2020-05-29:
Related topic: https://github.com/ARMmbed/mbed-os/pull/12838

adbridge, 2020-06-01:
@jeromecoutant "NA Master branch mbed-os-99.99.99" , this is why the bot is not accepting this issue for mirroring. It expects a valid sha or release tag.

jeromecoutant, 2020-06-01:
> "NA Master branch mbed-os-99.99.99" , this is why the bot is not accepting this issue for mirroring. It expects a valid sha or release tag.

See template:
Master branch is indicated by 'mbed-os-99.99.99

0xc0170, 2020-06-24:
>Related topic: #12838

This should be taken care of in this referenced PR. @jeromecoutant please review and add this there.

jeromecoutant, 2020-06-24:
> This should be taken care of in this referenced PR. @jeromecoutant please review and add this there.

Yes, but no...
Proposition was to remove LED use in greentea tests, whatever #12838 conclusion

0xc0170, 2020-06-25:
Understood, can you send a new pull request removing these 2 LEDs from there? not needed in the test

PR #13213 (2020-07-06): Reduce LED use in automatic CI tests
### Summary of changes 

Fix #13021 

LED use in automatic tests is not useful

- LED2 is removed
- LED1 is only kept in INTEGRATION tests as there is a way to not use it

#### Impact of changes 

none in CI

#### Migration actions required 

### Documentation 

----------------------------------------------------------------------------------------------------------------
### Pull request type 

    [x] Patch update (Bug fix / Target update / Docs update / Test update / Refactor)
    [] Feature update (New feature / Functionalit
closed 2020-07-06 by 0xc0170