ARMmbed/mbed-os #13023: STM32: Setting CAN bus speed on constructor doesn't work

FabianInostroza, 2020-05-27:
### Description of defect
Setting can bus speed on constructor doesn't work, the speed doesn't change.

#### Target(s) affected by this defect ?
Any STM32 with can bus

#### Toolchain(s) (name and version) displaying this defect ?
I'm using arm-none-eabi 9.2.1

#### What version of Mbed-os are you using (tag or sha) ?
mbed-os 5.15.3, branch mbed-os-5.15

The issue is caused by function `can_init_freq` that ignored the passed `hz` parameter:
https://github.com/ARMmbed/mbed-os/blob/f7861e6e4759b5a8cc4beb7d5ad1b5c40341ec6c/targets/TARGET_STM/can_api.c#L198
and
https://github.com/ARMmbed/mbed-os/blob/f7861e6e4759b5a8cc4beb7d5ad1b5c40341ec6c/targets/TARGET_STM/can_api.c#L695

this is already fixed in master branch.

jeromecoutant, 2020-05-28:
Hi
Yes, you should propose the same pull request as #12113 in mbed-os-5.15 branch

FabianInostroza, 2020-06-14:
> > you should propose the same pull request as #12113 in mbed-os-5.15 branch
> 
> @FabianInostroza did you succeed ?

Done

jeromecoutant, 2020-08-10:
Hi
Should be closed now #13114 has been merged ?

PR #13114 (2020-07-27): Fix issue #13023, CAN constructor ignores passed speed.

### Summary of changes 
Fixes #13023.

Uses the passed speed to the constructor.

#### Impact of changes 

#### Migration actions required
commit cf594418 (2020-07-27): Constructor ignored passed speed.
Fixes #13023.
closed 2020-08-20 by adbridge